innoveit/react-native-ble-manager #1230: Cant find device

MichauCorp, 2024-06-09:
Hello
This is not as much an issue as it is a request of assistance.

I am trying to create a simple react native app which scans and displays any BLE devices.
my ble device is an arduino type device - seeed xiao nrf52840.

the problem is I cant seem to find it (at least according to my understanding) when I use the scan function.
since im new to all this it might be an error in the display code however if possible it would mean a lot for me if you guys would take a look and perhaps give some insights.

thx in advance :)

`import React, { useEffect, useState } from 'react';
import { View, Text, FlatList, Button, PermissionsAndroid } from 'react-native';
import BleManager from 'react-native-ble-manager';

const BleScanner: React.FC = () => {
  const [bleDevices, setBleDevices] = useState<string[]>([]);

  useEffect(() => {
    requestPermissions();
    BleManager.enableBluetooth()
    BleManager.start({ showAlert: false });
    return () => {
      BleManager.stopScan();
    };
  }, []);
  
  /*
  const requestPermissions = async () => {
    try {
      const granted = await PermissionsAndroid.requestMultiple([
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADMIN,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
      ]);
      if (
        granted['android.permission.ACCESS_FINE_LOCATION'] === PermissionsAndroid.RESULTS.GRANTED &&
        granted['android.permission.BLUETOOTH_CONNECT'] === PermissionsAndroid.RESULTS.GRANTED &&
        granted['android.permission.BLUETOOTH_SCAN'] === PermissionsAndroid.RESULTS.GRANTED
      ) {
        console.log('All permissions granted');
        startScan();
      } else {
        console.log('Permission denied');
      }
    } catch (err) {
      console.warn(err);
    }
  };
  */

  const requestPermissions = async () => {
    try {
      
      const grantedLocation = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION
      );
      const grantedBluetoothConnect = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
      );
      const grantedBluetoothAdvertise = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADVERTISE
      );
      const grantedBluetoothScan = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN
      );
      /*
      const grantedLocation = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION
      );
      const grantedBluetoothAdmin = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADMIN
      );         
      const grantedBluetooth = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH
      );
      const grantedBluetoothConnect = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
      );
      const grantedBluetoothAdvertise = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADVERTISE
      );
      const grantedBluetoothCoarseLocation = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_COARSE_LOCATION
      );
      
      const grantedBluetoothScan = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN
      );
      */
      if (
        grantedLocation === PermissionsAndroid.RESULTS.GRANTED &&
        grantedBluetoothConnect === PermissionsAndroid.RESULTS.GRANTED &&
        //grantedBluetoothAdvertise === PermissionsAndroid.RESULTS.GRANTED &&
        grantedBluetoothScan === PermissionsAndroid.RESULTS.GRANTED
        /*
        grantedLocation === PermissionsAndroid.RESULTS.GRANTED &&
        grantedBluetoothAdmin === PermissionsAndroid.RESULTS.GRANTED 
        grantedBluetooth === PermissionsAndroid.RESULTS.GRANTED &&
        grantedBluetoothConnect === PermissionsAndroid.RESULTS.GRANTED &&
        grantedBluetoothAdvertise === PermissionsAndroid.RESULTS.GRANTED &&
        grantedBluetoothCoarseLocation === PermissionsAndroid.RESULTS.GRANTED &&
        grantedBluetoothScan === PermissionsAndroid.RESULTS.GRANTED
        */
      ) {
        console.log('All permissions granted');
        startScan();
      } else {
        console.log('Permission denied');
      }
    } catch (err) {
      console.warn(err);
    }
  };
  

  const startScan = () => {
    BleManager.scan([], 10, true).then(() => {
      console.log('Scan Started');
      console.log(bleDevices[0])
    });
  };

  const renderItem = ({ item }: { item: string }) => {
    return (
      <View style={{ padding: 10 }}>
        <Text>{item}</Text>
      </View>
    );
  };

  return (
    <View style={{ flex: 1, padding: 20 }}>
      <Button title="Start Scan" onPress={startScan} />
      <FlatList
        data={bleDevices}
        renderItem={renderItem}
        keyExtractor={(item, index) => index.toString()}
      />
    </View>
  );
};

export default BleScanner;
` 

note: there are a lot of notations because experimentation.

mrshahzeb7, 2024-06-09:
I am using "react-native-ble-plx"
and facing same issue "devices are not returning"

I am on react native 0.74.1 version 
I think issue might be related to latest version!

MichauCorp, 2024-06-13:
thank you for the comment @mrshahzeb7 

i tried implementing the ble-plx plugin before, and didnt manage because of some dependency issues.

do you happen to have a sample of your code by any chance?

jtovarto, 2024-06-20:
Hello.
Honestly, I didn't check the code you posted, I'll assume all permissions have been set and your logic is fine. If so, is your device connected to any other devices? Most of the time I faced this problem was when the device I was trying to find was already connected to another, this makes it impossible for the library to find it, maybe this can help you in some way.

closed 2024-07-24 by marcosinigaglia

strawhatmaceo, 2024-12-19:
How was this fixed?

jtovarto, 2024-12-19:
> How was this fixed?

The scan method only starts the device scanning process, but does not get the discovered devices, for them you have to implement the onDiscoverPeriphera listener, this is where you could get the information of the devices that are being discovered.