innoveit/react-native-ble-manager #1232: iOS read returns for wrong device

niklavsBariss, 2024-06-11:
When waking up from a screen lock, and reading from many (7 in my case) devices, return ByteArrays may be mixed up. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to device list
2. Lock screen
3. Wait a while
4. Unlock (this simultaneous starts read jobs for all devices) 
5. _sometimes_ wrong bytes are received

**Smartphone (please complete the following information):**
 - Device: iPhone 11 Pro Max
 - OS: 17.4.1
 - react-native-ble-manager version: 11.5.0
 - react-native version: 0.74.1

**Additional context**
We only detected this because we have multiple device types - ByteArrays of which are parsed differently, but they all share the same service and characteristic UUID (they all measure an environmental state (co2, temperature) of some sort). Reads are done in-parallel.

Does not happen on android so I'm guessing `readCallbacks` dictionary is to blame 

**Logs**
```
 LOG  receive TypeA [3, 0, 88, 2, 128, 0, 0, 252, 1, 38, 39, 65, 1, 3, 0, 0, 0, 3, 128, 81, 1, 0, 12, 0, 0, 0, 128, 58, 9, 0, 18, 0, 0, 0, 0, 141, 39, 0, 128, 105, 22, 255, 0, 0, 0, 0, 0]
 LOG  receive TypeB [3, 0, 88, 2, 128, 0, 0, 252, 1, 38, 39, 65, 1, 3, 0, 0, 0, 3, 128, 81, 1, 0, 12, 0, 0, 0, 128, 58, 9, 0, 18, 0, 0, 0, 0, 141, 39, 0, 128, 105, 22, 255, 0, 0, 0, 0, 0]
 LOG  GOT READINGS from 4A1755FA-F8DA-443E-E6F7-9D71FF61C99E (Type A)
 LOG  Error getting readings: invalid TypeB bytes 3,0,88,2,128,0,0,252,1,38,39,65,1,3,0,0,0,3,128,81,1,0,12,0,0,0,128,58,9,0,18,0,0,0,0,141,39,0,128,105,22,255,0,0,0,0,0
```
We can detect this (and read again) because TypeB would never have 3 in the first byte, but we can not detect if results of the same device type would get mixed up.

niklavsBariss, 2024-06-11:
Just checked and it happens on react-native-ble-manager version 1.5.3 too

niklavsBariss, 2024-06-11:
Just checked and it happens on react-native-ble-manager version 11.5.3 too

gotcha84, 2024-06-17:
This is likely the same issue fixed by https://github.com/innoveit/react-native-ble-manager/pull/1237.

closed 2024-06-18 by marcosinigaglia