innoveit/react-native-ble-manager #1240: Can't identify device ( Name is null for all devices )

aliwaqar981, 2024-06-20:
**Describe the bug**
Name and Local name is null for all scanned devices

**To Reproduce**
Steps to reproduce the behavior:
1. Just run the example given in the repo

**Expected behavior**
Name should be available for most of the devices.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Vivo v21e
 - OS: Android12
 - react-native-ble-manager version: ^11.5.5
 - react-native version: 0.74.2

**Additional context**
Mac address of device I'm trying to find is like 00:9C:C0:33:XX:XX but I'm not getting this while scanning through the example app. Although it's available when search from native bluetooth app of my phone.

aliwaqar981, 2024-06-23:
@typester @seigel @quatrix @Volcore any help would be appreciated.

varun761, 2024-06-25:
You can add a check if the device name is null don't display that. It will solve the issue that you are facing now.

aliwaqar981, 2024-06-26:
name is null for all devices. If I place this check then there won't be any device left

andreiwow2, 2024-07-05:
I think I might have the same issue, there is one device that I can find in the phone bluetooth list, but using react-native-ble-manager, not. On iOS it does work. The android permissions are correct, I can find other BLE devices, just this one not. So I assume it might find it but with a null name

brianwachira, 2024-07-23:
I am also experiencing the same issue but with another library. ( react-native-ble-plx ) 
I came to issues just to see whether this lib has the same issues

brianwachira, 2024-07-24:
> I am also experiencing the same issue but with another library. ( react-native-ble-plx ) I came to issues just to see whether this lib has the same issues

Update,  I ended up using React Native Bluetooth Classic. [https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-classic/](https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-classic/) In my use case, I am expected to send data between mobile devices & device name is needed ( inorder to identify device ). It seems Bluetooth low energy only allows smallest bytes of device information to be sent, hence name may miss at times

closed 2024-07-24 by marcosinigaglia

timmyjose, 2024-08-27:
> Update, I ended up using React Native Bluetooth Classic. https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-classic/ In my use case, I am expected to send data between [...]

Hey @brianwachira, just curious -  I too have a usecase where I need to communicate between devices (`iOS` and `Android`). Did you manage to get the communication done using the library you linked to? If you have some demo app or pointers, that'd be great!