innoveit/react-native-ble-manager #1246: getConnectedPeripherals does not populate the advertising data on Android

Jetbo, 2024-07-11:
**Describe the bug**

Hello, I think I ended up finding an oversight in the implementation of `getConnectedPeripherals`. On Android, it seems the `advertising` data is never populated. See the code snippets below...

iOS, correctly attaches the `advertising` info here: https://github.com/innoveit/react-native-ble-manager/blob/d7613a48ff201914f4a91d7167f35d615ce0d24a/ios/BleManager.swift#L519

Android, does not: https://github.com/innoveit/react-native-ble-manager/blob/d7613a48ff201914f4a91d7167f35d615ce0d24a/android/src/main/java/it/innove/BleManager.java#L744-L761

From the best I can tell, the only way to get that information back on Android is via ScanResult.getScanRecord() like you have here: https://github.com/innoveit/react-native-ble-manager/blob/d7613a48ff201914f4a91d7167f35d615ce0d24a/android/src/main/java/it/innove/DefaultScanManager.java#L197C36-L197C49

**Expected behavior**

At the very least, I think the documentation should be updated for Android that advertising data cannot be read without first having a ScanResult.

GiridharKarnik, 2024-09-19:
@Jetbo , were you able to read the advertising data at all?

lucaswitch, 2024-09-20:
> **Describe the bug**
> 
> Hello, I think I ended up finding an oversight in the implementation of `getConnectedPeripherals`. On Android, it seems the `advertising` data is never populated. See the code snippets below...
> 
> iOS, correctly attaches the `advertising` info here:
> 
> https://github.com/innoveit/react-native-ble-manager/blob/d7613a48ff201914f4a91d7167f35d615ce0d24a/ios/BleManager.swift#L519
> 
> Android, does not:
> 
> https://github.com/innoveit/react-native-ble-manager/blob/d7613a48ff201914f4a91d7167f35d615ce0d24a/android/src/main/java/it/innove/BleManager.java#L744-L761
> 
> From the best I can tell, the only way to get that information back on Android is via `ScanResult` like you have here:
> 
> https://github.com/innoveit/react-native-ble-manager/blob/d7613a48ff201914f4a91d7167f35d615ce0d24a/android/src/main/java/it/innove/DefaultScanManager.java#L197
> 
> **Expected behavior**
> 
> At the very least, I think the documentation should be updated for Android that advertising data cannot be read without first having a ScanResult.

@Jetbo 
Maybe i'm wrong but i guess Android only shows that information during scanning of nearby devices.
But you're right the getConnectedDevices() method never returns adv info maybe is something that we can update on the docs to justify that behavior and close this issue.

lucaswitch, 2024-09-22:
> **Describe the bug**
>
> Hello, I think I ended up finding an oversight in the implementation of `getConnectedPeripherals`. On Android, it seems the `advertising` data is never populated. See the code [...]

@Jetbo addressed improving docs.

closed 2024-09-25 by Jetbo