innoveit/react-native-ble-manager #1253: Connection Priority

thompsonnoah, 2024-08-26:
Hello!

I'd like to make a feature request. Currently, `requestConnectionPriority` is only available to Android smartphones. I'd like to request that it would be made available to iOS devices using the `setDesiredConnectionLatency` function available [here.](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/setdesiredconnectionlatency(_:for:)) It looks like an alternative library was able to do it, as per [this issue](https://github.com/dotintent/react-native-ble-plx/issues/134). Therefore, I'd like to request that it's available here, as react-native-plx didn't work for me.

**Smartphone:**
 - Device:iPhone 13
 - OS: 17.6.1
 - react-native-ble-manager version: 11.5.5
 - react-native version: 0.74.3

**Additional context**
I need to send the data from my peripheral at a very fast rate (more than 60Hz).

marcosinigaglia, 2024-08-27:
Hi, the setDesiredConnectionLatency function is only for Bluetooth Classic and not for BLE so it is impossible to implement it.

closed 2024-08-27 by marcosinigaglia

matheusjardimb, 2024-11-11:
Hello there! 

@thompsonnoah, have you found any solution for the issue you pointed?

@marcosinigaglia, thanks for your support! Have you considered @thompsonnoah's issue (https://github.com/dotintent/react-native-ble-plx/issues/134#issuecomment-1750188203)? It seems they've had some progress on the subject https://github.com/search?q=repo%3Adotintent%2Freact-native-ble-plx+ConnectionPriority&type=code

thompsonnoah, 2024-11-12:
Hi @matheusjardimb,

I wasn't able to find a solution to this issue, it appears that with `react-native-ble-manager`, this feature is unavailable. I went with the default settings and it worked okay for me by reducing the packet size.

I'm also curious if this feature could still be implemented in this library, as `react-native-ble-plx` was able to include it.

Thanks,

Noah

marcosinigaglia, 2024-11-15:
Hi, I don't see any implementation on iOS.

thompsonnoah, 2024-11-21:
Hi @marcosinigaglia,

Thanks for the reply! I think the `react-native-ble-plx` team was able to include the Swift code in their repo, as shown here: [https://github.com/search?q=repo%3Adotintent%2Freact-native-ble-plx+ConnectionPriority&type=code#:~:text=6%20more%20matches-,ios/MultiplatformBleAdapter/BleAdapter.swift,-Swift](https://github.com/search?q=repo%3Adotintent%2Freact-native-ble-plx+ConnectionPriority&type=code#:~:text=6%20more%20matches-,ios/MultiplatformBleAdapter/BleAdapter.swift,-Swift)

Definitely let me know your thoughts!

Best,

Noah

marcosinigaglia, 2024-11-21:
Hi @thompsonnoah , I checked again the code and the function is doing nothing.