innoveit/react-native-ble-manager #1257: Cannot read property 'start' of null

medero, 2024-09-10:
**Describe the bug**
When initiating BleManager.start I get 'cannot read property 'start' of null in Expo 51 / React Native 0.74.5 and thus unable to get permissions and scan devices (my goal is connecting to an Aranet4 device)

Implementation here: https://github.com/medero/react-native-ble-manager-expo-example/blob/main/app/(tabs)/index.tsx

**To Reproduce**
Steps to reproduce the behavior:
1. Clone this repository: https://github.com/medero/react-native-ble-manager-expo-example
2. yarn & yarn run android
3. Scan the QR code (in my case, Galaxy S22) using this development mode (is this the same as using EOS?)
4. Experience the error "Cannot read property 'start' of null"

**Expected behavior**
Expectation is to not receive that error in order to establish bluetooth permissions and scan for devices.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy 23
 - OS: Android 13
 - react-native-ble-manager version: ^11.5.5
 - react-native version: 0.74.5

marcosinigaglia, 2024-09-10:
You have to use a development build https://docs.expo.dev/develop/development-builds/create-a-build/

medero, 2024-09-10:
Ah - so I was confusing Expo Go's development mode with a proper eos development build?

medero, 2024-09-10:
> You have to use a development build https://docs.expo.dev/develop/development-builds/create-a-build/

Any tips on speeding up eos builds?

medero, 2024-09-10:
> You have to use a development build https://docs.expo.dev/develop/development-builds/create-a-build/

I updated the repo after making an EOS build and installing the app on my Phone and still no dice..

INFO  never_ask_again
 INFO  never_ask_again
 INFO  never_ask_again
 INFO  request android 31
 INFO  never_ask_again
 INFO  never_ask_again
 INFO  never_ask_again
 WARN  Bluetooth permissions not granted.

Note: In my repo I nullified app.json's eas.projectId. 

Screenshot of the app:

[image]

medero, 2024-09-11:
I removed my initial call to requestPermissions on HomeScreen load and just am calling it on the Scan button Press.. I actually ended up getting this easily to work with react-native-ble-plx which had app.json with permissions set in android.permissions, tried to emulate it and no dice still. 

Working react-native-ble-plx https://github.com/medero/react-native-ble-plx-example - it also uses the @config-plugins/react-native-ble-plx which seems like it could be the trick to why this works: 

        "@config-plugins/react-native-ble-plx",
        {
          "isBackgroundEnabled": true,
          "modes": [
            "peripheral",
            "central"
          ],
          "bluetoothAlwaysPermission": "Allow $(PRODUCT_NAME) to connect to bluetooth devices"

Would be curious to see if I missed anything with ble-manager here.

medero, 2024-09-11:
This is probably why https://blog.theodo.com/2023/07/ble-integration-for-rn-apps/ mentions it doesn't have support for Expo

marcosinigaglia, 2024-09-12:
@medero the example in the repo use expo.

closed 2024-09-12 by marcosinigaglia