innoveit/react-native-ble-manager #1273: Can't find devices

looooool11111, 2024-10-09:
**Describe the bug**
Bluetooth devices cannot be searched on Android 11 devices, but can be used normally on devices larger than Android 11

**Screenshots**
Screenshots
If applicable, add screenshots to help explain your problem.
AndroidManifest.xml
[image]

startScan
[image]

Can't find devices
[image]

init
[image]

**Smartphone (please complete the following information):**
 - OS: [e.g. iOS8.1]:android 11
 - react-native-ble-manager:11.5.5
 - react-native version:0.73.4

lucaswitch, 2024-10-09:
Thats actually a problem with runtime permissions on Android. I can confirm that is working as expected on Android 11.

looooool11111, 2024-10-10:
> Thats actually a problem with runtime permissions on Android. I can confirm that is working as expected on Android 11.

yeah,I checked the device's permission settings and found that my application has obtained Bluetooth related permissions, but has not requested to use them

lucaswitch, 2024-10-10:
> > Thats actually a problem with runtime permissions on Android. I can confirm that is working as expected on Android 11.
> 
> yeah,I checked the device's permission settings and found that my application has obtained Bluetooth related permissions, but has not requested to use them

Did you was able to solve it?

looooool11111, 2024-10-11:
> Did you was able to solve it?

not yet