innoveit/react-native-ble-manager #1274: Not compatible with default architecture 0.75.4 on IOS

lucaswitch, 2024-10-09:
### Crash on trying to add the library on a brand 0.75.4 react-native app.

Just trying the library on bare react-native app with latest version.
It builds but crashes as soon ble-manager is added through js bindings on the app.

Details:
[image]

What we know:
Looks like interop works for most of ble-manager methods as we can see on "methods" property on the stack trace.
27 of 34 methods are being correctly exported to the new arch. 
Probably we need to investigate further why that happens with interop layer and react-native-ble-manager.

Also we all know this is happening because since react-native 0.74, **bridge was removed** but since react-native team added a interop layer for backward compability as discussed here: https://github.com/reactwg/react-native-new-architecture/discussions/175 to support most of used libraries without having to changing the current scope of those libraries.

[image]

It still works with **bridgeless disabled**.
But i think we need to make the adjusts to support bridgeless interop layer(first) then actually add the new turbo methods as discussed here: https://github.com/reactwg/react-native-new-architecture/blob/main/docs/backwards-compat-turbo-modules.md

Full XCode stacktrace:
`#0	0x00000001fa607dc4 in _platform_strlen ()
#1	0x00000001009033fc in std::__1::__constexpr_strlen[abi:v160006](char const*) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/cstring:114
#2	0x00000001009032fc in std::__1::char_traits<char>::length(char const*) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__string/char_traits.h:220
#3	0x0000000100903280 in std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string[abi:v160006]<std::nullptr_t>(char const*) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/string:882
#4	0x0000000100902124 in std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string[abi:v160006]<std::nullptr_t>(char const*) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/string:880
#5	0x000000010119c628 in facebook::react::(anonymous namespace)::parseExportedMethods(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, objc_class*) at /Users/lucas/Documents/impeto/metabolic-app/node_modules/react-native/ReactCommon/react/nativemodule/core/platform/ios/ReactCommon/RCTInteropTurboModule.mm:126
#6	0x000000010119bd24 in facebook::react::ObjCInteropTurboModule::ObjCInteropTurboModule(facebook::react::ObjCTurboModule::InitParams const&) at /Users/lucas/Documents/impeto/metabolic-app/node_modules/react-native/ReactCommon/react/nativemodule/core/platform/ios/ReactCommon/RCTInteropTurboModule.mm:188
#7	0x000000010119d5fc in facebook::react::ObjCInteropTurboModule::ObjCInteropTurboModule(facebook::react::ObjCTurboModule::InitParams const&) at /Users/lucas/Documents/impeto/metabolic-app/node_modules/react-native/ReactCommon/react/nativemodule/core/platform/ios/ReactCommon/RCTInteropTurboModule.mm:187
#8	0x00000001011d6ab8 in facebook::react::ObjCInteropTurboModule* std::__1::construct_at[abi:v160006]<facebook::react::ObjCInteropTurboModule, facebook::react::ObjCTurboModule::InitParams&, facebook::react::ObjCInteropTurboModule*>(facebook::react::ObjCInteropTurboModule*, facebook::react::ObjCTurboModule::InitParams&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__memory/construct_at.h:38
#9	0x00000001011d68d8 in void std::__1::allocator_traits<std::__1::allocator<facebook::react::ObjCInteropTurboModule>>::construct[abi:v160006]<facebook::react::ObjCInteropTurboModule, facebook::react::ObjCTurboModule::InitParams&, void, void>(std::__1::allocator<facebook::react::ObjCInteropTurboModule>&, facebook::react::ObjCInteropTurboModule*, facebook::react::ObjCTurboModule::InitParams&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__memory/allocator_traits.h:304
#10	0x00000001011d6804 in std::__1::__shared_ptr_emplace<facebook::react::ObjCInteropTurboModule, std::__1::allocator<facebook::react::ObjCInteropTurboModule>>::__shared_ptr_emplace[abi:v160006]<facebook::react::ObjCTurboModule::InitParams&>(std::__1::allocator<facebook::react::ObjCInteropTurboModule>, facebook::react::ObjCTurboModule::InitParams&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__memory/shared_ptr.h:284
#11	0x00000001011d64f0 in std::__1::__shared_ptr_emplace<facebook::react::ObjCInteropTurboModule, std::__1::allocator<facebook::react::ObjCInteropTurboModule>>::__shared_ptr_emplace[abi:v160006]<facebook::react::ObjCTurboModule::InitParams&>(std::__1::allocator<facebook::react::ObjCInteropTurboModule>, facebook::react::ObjCTurboModule::InitParams&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__memory/shared_ptr.h:276
#12	0x00000001011d63f8 in std::__1::shared_ptr<facebook::react::ObjCInteropTurboModule> std::__1::allocate_shared[abi:v160006]<facebook::react::ObjCInteropTurboModule, std::__1::allocator<facebook::react::ObjCInteropTurboModule>, facebook::react::ObjCTurboModule::InitParams&, void>(std::__1::allocator<facebook::react::ObjCInteropTurboModule> const&, facebook::react::ObjCTurboModule::InitParams&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__memory/shared_ptr.h:995
#13	0x00000001011cb710 in std::__1::shared_ptr<facebook::react::ObjCInteropTurboModule> std::__1::make_shared[abi:v160006]<facebook::react::ObjCInteropTurboModule, facebook::react::ObjCTurboModule::InitParams&, void>(facebook::react::ObjCTurboModule::InitParams&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__memory/shared_ptr.h:1004
#14	0x00000001011cb48c in -[RCTTurboModuleManager provideLegacyModule:] at /Users/lucas/Documents/impeto/metabolic-app/node_modules/react-native/ReactCommon/react/nativemodule/core/platform/ios/ReactCommon/RCTTurboModuleManager.mm:470
#15	0x00000001011dc3bc in -[RCTTurboModuleManager installJSBindings:]::$_1::operator()(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) const at /Users/lucas/Documents/impeto/metabolic-app/node_modules/react-native/ReactCommon/react/nativemodule/core/platform/ios/ReactCommon/RCTTurboModuleManager.mm:959
#16	0x00000001011dc330 in decltype(std::declval<-[RCTTurboModuleManager installJSBindings:]::$_1&>()(std::declval<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&>())) std::__1::__invoke[abi:v160006]<-[RCTTurboModuleManager installJSBindings:]::$_1&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&>(-[RCTTurboModuleManager installJSBindings:]::$_1&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__functional/invoke.h:394
#17	0x00000001011dc2d8 in std::__1::shared_ptr<facebook::react::TurboModule> std::__1::__invoke_void_return_wrapper<std::__1::shared_ptr<facebook::react::TurboModule>, false>::__call<-[RCTTurboModuleManager installJSBindings:]::$_1&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&>(-[RCTTurboModuleManager installJSBindings:]::$_1&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) at /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS17.2.sdk/usr/include/c++/v1/__functional/invoke.h:478
#18	0x00000001011dc2a4 in std::__1::__function::__alloc_func<-[RCTTurboModuleManager installJSBindings:]::$_1, std::__1::allocator<-[RCTTurboModuleManager installJSBindings:]::$_1>, std::__1::shared_ptr<facebook::react::TurboModule> (std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&)>::operato [... cut ...]
#94	0x000000019d8eec9c in __CFRUNLOOP_IS_CALLING_OUT_TO_A_BLOCK__ ()
#95	0x000000019d8dcdec in __CFRunLoopDoBlocks ()
#96	0x000000019d8dcad0 in __CFRunLoopRun ()
#97	0x000000019d8dbcd8 in CFRunLoopRunSpecific ()
#98	0x00000001013a22a0 in +[RCTJSThreadManager runRunLoop] at /Users/lucas/Documents/impeto/metabolic-app/node_modules/react-native/ReactCommon/react/runtime/platform/ios/ReactCommon/RCTJSThreadManager.mm:102
#99	0x000000019c813718 in __NSThread__start__ ()
#100	0x00000001fa6c006c in _pthread_start ()
`

thompsonnoah, 2024-10-25:
Hello!

I'm having the same issue after enabling the New Architecture in my app and upgrading to 0.76. The app starts, then crashes after the BleManager is created. @marcosinigaglia Is there any ETA on an update given that the new arch is enabled by default?

Thanks,

Noah

lucaswitch, 2024-10-25:
I'm working on that on my little spare time(and not tired) @thompsonnoah 
I already did half of the work to migrate that to new arch as you can see the progress on my personal branch https://github.com/lucaswitch/react-native-ble-manager

It is still working in progress but mostly because the new codegen needs the explicit type definitions of each "object" in methods which needs a little more attention to be used.

What needs to be done:
1) Each ts "object" needs to be properly explicit typed accordingly with the docs and the old bridge methods.
2) Also i need to add an new arch app to the repo so we can control if something goes wrong on new arch methods while we can keep the bridge methods.

Probably an 12.x candidate.
[[See the codegen file](https://github.com/lucaswitch/react-native-ble-manager/blob/master/src/NativeBleManager.ts)](https://github.com/lucaswitch/react-native-ble-manager/blob/master/src/NativeBleManager.ts)

thompsonnoah, 2024-10-25:
Hi @lucaswitch,

Thanks so much for the update! I'll be actively monitoring this post for news.

Best,

Noah

lucaswitch, 2024-10-29:
@thompsonnoah @marcosinigaglia i think i got it working, still needs to be fully tested.
I'll try it on a production app then i guess we have a 12.x.x

thompsonnoah, 2024-10-29:
Hi @lucaswitch,

Thanks so much for the amazing effort! I look forward to hearing more and to the 12.x update.

Best,

Noah

lucaswitch, 2024-10-30:
It's not quite ready yet. I guess it needs some work to make the events compatible again.

vhakulinen, 2024-11-02:
I had a peek at your branch. Did I understand correctly that the codegen generated from the types will call into the current native methods, and that is the provided interop for legacy modules?

To make the typing more robust, would it be possible that the "actual" `BleManager` in typescript side implements the `Spec` interface?

marcosinigaglia, 2024-11-04:
Hi @lucaswitch , thanks for your work, can you make a PR so we can work on your code?

lucaswitch, 2024-11-04:
> I had a peek at your branch. Did I understand correctly that the code generated from the types will call into the current native methods, and that is the provided interop for legacy modules?

Yes, you're right.
But not with the old RCT_* macros, thats the old way to do it. But that should be kept for backwards compability.
The typescript will do the turbo modules bindings when user install pods.

> To make the typing more robust, would it be possible that the "actual" `BleManager` in typescript side implements the `Spec` interface?

Thats a great idea.

lucaswitch, 2024-11-04:
> Hi @lucaswitch , thanks for your work, can you make a PR so we can work on your code?

I'll do that right now.

EricChou517, 2024-11-07:
@lucaswitch Sorry , My app show EventEmitter Type not found ?

lucaswitch, 2024-11-07:
> @lucaswitch Sorry , My app show EventEmitter Type not found ?

Hey @EricChou517, that branch is still under development. Don't try to use it yet.

lucaswitch, 2024-12-06:
This was fixed on version 12.0.0-rc.3, update your branches with this rc version to enabled new arch.

closed 2024-12-09 by marcosinigaglia