innoveit/react-native-ble-manager #1276: Android crash when cancel scanCompanion

francesco-clementi-92, 2024-10-21:
I configured all the required permissions and read carefully the documentation.

When I invoke BleManager.companionScan([]) a window popup appear in order to select a bluetooth device. If I click outside the popup, the app will crash without any logs on JS side.

Logs on android
```
2024-10-20 17:45:12.970  1858-1858  AlarmManager            system_server                        D  mBroadcastRefCount -> 6
2024-10-20 17:45:12.970  1858-1858  AlarmManager            system_server                        D  mPendingIntent -> null
2024-10-20 17:45:12.973  1858-1858  AlarmManager            system_server                        D  mBroadcastRefCount -> 5
2024-10-20 17:45:12.973  1858-1858  AlarmManager            system_server                        D  mPendingIntent -> null
2024-10-20 17:45:12.974  1858-1858  AlarmManager            system_server                        D  mBroadcastRefCount -> 4
2024-10-20 17:45:12.974  1858-1858  AlarmManager            system_server                        D  mPendingIntent -> null
2024-10-20 17:45:12.975  1858-1858  AlarmManager            system_server                        D  mBroadcastRefCount -> 3
2024-10-20 17:45:12.975  1858-1858  AlarmManager            system_server                        D  mPendingIntent -> null
2024-10-20 17:45:12.975  1858-1858  AlarmManager            system_server                        D  mBroadcastRefCount -> 2
2024-10-20 17:45:12.975  1858-1858  AlarmManager            system_server                        D  mPendingIntent -> PendingIntent{5641f87: PendingIntentRecord{7dc8ff4 com.miui.securitycenter startService}}
2024-10-20 17:45:12.975  1858-1858  AlarmManager            system_server                        D  mBroadcastRefCount -> 1
2024-10-20 17:45:12.976  1858-1858  AlarmManager            system_server                        D  mPendingIntent -> PendingIntent{6a6beb4: PendingIntentRecord{f946575 com.google.android.gms/com.google.android.gms.scheduler broadcastIntent}}
2024-10-20 17:45:12.976  1858-1858  AlarmManager            system_server                        D  mBroadcastRefCount -> 0
2024-10-20 17:45:12.976  1858-1858  AlarmManager            system_server                        D  mPendingIntent -> PendingIntent{5083efe: PendingIntentRecord{e28665f com.miui.powerkeeper startService}}
2024-10-20 17:45:12.984  3267-3267  ReflectUtils            com.android.systemui                 E  invokeObject
                                                                                                    java.lang.reflect.InvocationTargetException
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.miui.clock.utils.ReflectUtils.invokeObject(ReflectUtils.java:31)
                                                                                                    	at com.miui.clock.utils.DataUtils.isAppInstalledForUser(DataUtils.java:341)
                                                                                                    	at com.miui.clock.utils.DataUtils.isHealthAppInstalled(DataUtils.java:326)
                                                                                                    	at com.miui.clock.classic.ClassicClockView.filterClassicClockInfo(ClassicClockView.java:178)
                                                                                                    	at com.miui.clock.classic.ClassicClockView.updateTime(ClassicClockView.java:316)
                                                                                                    	at com.miui.clock.MiuiClockController$1.run(MiuiClockController.java:155)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:211)
                                                                                                    	at android.os.Looper.loop(Looper.java:300)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8503)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:561)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:954)
                                                                                                    Caused by: android.content.pm.PackageManager$NameNotFoundException: com.mi.health
                                                                                                    	at android.app.ApplicationPackageManager.getPackageInfoAsUser(ApplicationPackageManager.java:254)
                                                                                                    	at android.app.ApplicationPackageManager.getPackageInfoAsUser(ApplicationPackageManager.java:242)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.miui.clock.utils.ReflectUtils.invokeObject(ReflectUtils.java:31) 
                                                                                                    	at com.miui.clock.utils.DataUtils.isAppInstalledForUser(DataUtils.java:341) 
                                                                                                    	at com.miui.clock.utils.DataUtils.isHealthAppInstalled(DataUtils.java:326) 
                                                                                                    	at com.miui.clock.classic.ClassicClockView.filterClassicClockInfo(ClassicClockView.java:178) 
                                                                                                    	at com.miui.clock.classic.ClassicClockView.updateTime(ClassicClockView.java:316) 
                                                                                                    	at com.miui.clock.MiuiClockController$1.run(MiuiClockController.java:155) 
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99) 
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:211) 
                                                                                                    	at android.os.Looper.loop(Looper.java:300) 
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8503) 
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:561) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:954) 
2024-10-20 17:45:12.986  3267-3267  ReflectUtils            com.android.systemui                 E  invokeObject
                                                                                                    java.lang.reflect.InvocationTargetException
[... cut ...]
2024-10-20 17:45:12.990  1858-1858  Compatibil...geReporter system_server                        D  Compat change id reported: 135634846; UID 6101; state: DISABLED
2024-10-20 17:45:12.991  1858-1858  Compatibil...geReporter system_server                        D  Compat change id reported: 177438394; UID 6101; state: DISABLED
1970-01-01 01:00:00.000     0-0     <no-tag>                                                     I  logcat: Unexpected EOF!
```

 - Device: Xiaomi Redmi Note 11
 - OS: Android 13
 - react-native-ble-manager version:  11.5.6
 - react-native version: 0.75.4

vhakulinen, 2024-10-30:
Doesn't really point to this package: `Caused by: android.content.pm.PackageManager$NameNotFoundException: com.mi.health` and `Caused by: android.content.pm.PackageManager$NameNotFoundException: com.miui.weather2`.

Needs more information.

vhakulinen, 2024-11-01:
And the "more information" is here:
https://github.com/innoveit/react-native-ble-manager/blob/326f579ff85ac22ed7934a11605d3d7eab1997ee/android/src/main/java/it/innove/CompanionScanner.java#L125-L126

That should have null check

PR #1281 (2024-11-04): bugfix: add missing null checks to CompanionScanner
Add missing null checks to companion scanner, and make sure that the activity result always ends up calling the scan callback.

Fixes #1276