innoveit/react-native-ble-manager #1278: v11.5.6 beacon scanning issue

devirtua, 2024-10-25:
On the v.11.5.6, for the 
`
  <uses-permission
    android:name="android.permission.BLUETOOTH_SCAN"
    android:usesPermissionFlags="neverForLocation"
    tools:targetApi="s" />
` permission,
`android:usesPermissionFlags="neverForLocation"`
flag prevents scanner from finding beacons even though adding
`
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
`
permission to main project.

Removing `android:usesPermissionFlags="neverForLocation"` flag solves the problem.

It took me a full day to resolve the issue.

lucaswitch, 2024-10-29:
Thats not a problem with the library itself but a environment configuration problem.
But we'll gonna add that into the docs so nobody suffers with that.

ShawnReardon, 2024-12-06:
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags" />

This resolved it for me

marcosinigaglia, 2024-12-06:
Added a note in the doc.

closed 2024-12-06 by marcosinigaglia

devirtua, 2024-12-06:
doc says `If you are working with Beacons remove the android:usesPermissionFlags="neverForLocation".` but it is not working, `tools:remove="android:usesPermissionFlags"` should be used

ShawnReardon, 2024-12-07:
> doc says `If you are working with Beacons remove the android:usesPermissionFlags="neverForLocation".` but it is not working, `tools:remove="android:usesPermissionFlags"` should be used

So a few things: If you (in trying to solve this) have something like react-native-ble-plx in your modules, they also have this set and the remove is needed unless you edit all the other manifest. 

With the neverForLocation removed, no other BLE libraries and this as my manifest its works:

`<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

<uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />

<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE"/>
<uses-permission android:name="android.permission.NEARBY_DEVICES"/>
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
`

That said I would personally prefer they remove it from the default Manifest because it sets a preference in an unclear way, in a way which tramples choices you think you are making elsewhere regarding being explicit about if/how you access location info.  Nothing about the library screams to me that it wouldn't work with Beacons so I find this a bit odd as a choice.