innoveit/react-native-ble-manager #1286: Can't find devices on IOS Scan

Andre-Capelli, 2024-11-05:
Hello, I'm a trouble with the IOS on scanning the devices. The start(), scan() all works fine, but the eventListener does not return any device on IOS only, Android works perfectly, here is what I have:

const BleManagerModule = NativeModules.BleManager;
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);

class W {
  btItems: BLE.Peripheral[] = [];

  startModuleBT = async () => {
    try {
      return new Promise((resolve, reject) =>
        BLE.start()
          .then((v) => resolve(v))
          .catch((e) => reject(e))
      );
    } catch (error) {}
  };

  startScanBT = async (
    model: string,
    setItems: React.Dispatch<React.SetStateAction<BLE.Peripheral[]>>
  ) => {
    try {
      await this.startModuleBT();
      this.btItems = [];

      BLE.scan([], 120, true)
        .then((v) => {
          return Promise.resolve(v);
        })
        .catch((e) => {
          console.error(e);
          return Promise.reject(e);
        });

      bleManagerEmitter.addListener(
        "BleManagerDiscoverPeripheral",
        (device: BLE.Peripheral) => {
          console.log(device);
          if (
            device &&
            device.name &&
            device.id &&
            device.name.includes(model.toUpperCase())
          ) {
            const exists = this.btItems.some((d) => d.id === device.id);

            if (!exists) {
              this.btItems.push(device);
              setItems(this.btItems);
            }
          }
        }
      );
    } catch (error) {}
  };
  ....}
  
  I have created a Context for the BLE start:
  
  export const BLEManagerContext = createContext({});

export const BLEManagerProvider = ({
  children,
}: {
  children: React.ReactNode;
}) => {
  useEffect(() => {
    BLE.start({
      showAlert: false,
      forceLegacy: true,
    })
      .then((v) => {
        console.log("STARTED ", Platform.OS);
        return Promise.resolve(v);
      })
      .catch((e) => {
        console.log("STARTED ERROR", Platform.OS);
        return Promise.reject(e);
      });
  }, []);

  return (
    <BLEManagerContext.Provider value={{}}>
      {children}
    </BLEManagerContext.Provider>
  );
};

export const useConnection = () => useContext(BLEManagerContext);

It's being called when the app starts, the IOS permissions for bluetooth are all declared and accepted:

	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>BT access to scan nearby devices </string>
	<key>NSBluetoothPeripheralUsageDescription</key>
	<string>BT access to scan nearby devices </string>
	<key>UIBackgroundModes</key>
	<array>
		<string>central-peripheral</string>
		<string>bluetooth-central</string>
		<string>processing</string>
		<string>remote-notification</string>
	</array>

I'm testing in a physical Iphone 13, OS 17.6.1
Package BLE Manager,  "react-native-ble-manager": "^11.5.6",

Andre-Capelli, 2024-11-05:
Edit:

Solved the problem by calling the listeners after the start module, and not with the scan method

closed 2024-11-05 by Andre-Capelli

sohaibbinmohsin, 2025-07-01:
Hello @Andre-Capelli, I too am facing the same issue. My iOS device doesnt discover any peripherals even though i can see bluetooth devices shown up in the device. Here's my code:
```
  const startRealScan = () => {
      let onDiscoverListener: any;
      addDebugLog('Starting real Bluetooth scan...');

      BleManager.start({showAlert: true}).then(() => {
        addDebugLog('BLE Manager started successfully');
      });

      BleManager.checkState().then(state =>
        console.log(`current BLE state = '${state}'.`),
      );
      // Register the callback for peripheral discovery
      onDiscoverListener = BleManager.onDiscoverPeripheral(
        (peripheral: any) => {
          addDebugLog(
            `Found: ${peripheral.name || 'Unknown'} (${peripheral.rssi} dBm)`,
          );

          // Determine device type based on advertising data
          const getDeviceType = (peripheral: any) => {
            const {advertising} = peripheral;

            // Check for specific service UUIDs that indicate device type
            if (advertising?.serviceUUIDs) {
              const services = advertising.serviceUUIDs.map((uuid: string) =>
                uuid.toLowerCase(),
              );

              // Common service UUIDs for different device types
              if (services.some((uuid: string) => uuid.includes('180d')))
                return 'Fitness Tracker';
              if (services.some((uuid: string) => uuid.includes('180f')))
                return 'Battery Service';
              if (services.some((uuid: string) => uuid.includes('1812')))
                return 'HID Device';
              if (services.some((uuid: string) => uuid.includes('110a')))
                return 'Audio Device';
              if (services.some((uuid: string) => uuid.includes('110e')))
                return 'Phone';
              if (services.some((uuid: string) => uuid.includes('110d')))
                return 'Computer';
            }

            // Check manufacturer data for device type hints
            if (advertising?.manufacturerData) {
              const manufacturerId = Object.keys(
                advertising.manufacturerData,
              )[0];
              // Apple devices
              if (manufacturerId === '76') return 'Apple Device';
              // Samsung devices
              if (manufacturerId === '15') return 'Samsung Device';
              // Google devices
              if (manufacturerId === '224') return 'Google Device';
            }

            // Check device name for hints
            const name = peripheral.name?.toLowerCase() || '';
            if (name.includes('iphone') || name.includes('ipad'))
              return 'Apple Device';
            if (name.includes('samsung')) return 'Samsung Device';
            if (name.includes('jsn') || name.includes('huawei'))
              return 'Huawei Device';
            if (name.includes('macbook') || name.includes('mac'))
              return 'Computer';
            if (name.includes('laptop')) return 'Computer';
            if (name.includes('watch')) return 'Smartwatch';
            if (name.includes('headphone') || name.includes('buds'))
[... 139 lines cut ...]
            'No Devices Found',
            'No Bluetooth devices were discovered during the scan. This could be because:\n\n• No devices are nearby\n• Bluetooth is disabled on nearby devices\n• Devices are not advertising\n• Location permission is required for Bluetooth scanning\n\nTry moving to a different location or ensure Bluetooth is enabled on nearby devices.',
            [
              {
                text: 'OK',
                onPress: () => navigation.navigate('MainTabs'),
              },
            ],
          );
        } else {
          addDebugLog(`✅ ${deviceCount} devices found successfully`);
          // Show success alert
          Alert.alert(
            'Scan Complete',
            `Successfully discovered ${deviceCount} Bluetooth device(s)!`,
            [{text: 'OK'}],
          );
        }
      }, 20000);

      setScanTimeout(timeout);
    };
```

Can you figure out what issue i have?

Doko-Demo-Doa, 2025-07-02:
Same issue. iOS 18 with iPhone 16 Pro.