innoveit/react-native-ble-manager #1287: Long response after request

SaNFeeDep, 2024-11-12:
The problem is that when sending commands to the device, the response time from the device is 10 seconds. 
I use Expo for assembly. I thought that when assembling the “product” version this time would become shorter, but it remained the same.
Below is an example of a request and response. Please pay attention to the time.

`LOG  BLEManager: Send command = "pass 0|who|fuel|bat|temp|time" on device C5:8B:00:78:C8:FD. Time: 13:37:38`
`LOG  BLEManager: Writed...`
`LOG  BLEManager: Received => PASS=OK|WHO=FW:0.0.1,SN:12340000|FUEL=0|BAT=3.1|TEMP=25|TIME=0,6345. Time: 13:37:48.`

It takes 10 seconds to respond to this command -_-

Below is the class I use to work with BleManager.

```
import { Buffer } from 'buffer'
import { Permission, PermissionsAndroid, Platform } from 'react-native'
import BleManager, {
  BleConnectPeripheralEvent,
  BleDisconnectPeripheralEvent,
  BleDiscoverPeripheralEvent,
  BleManagerDidUpdateStateEvent,
  BleManagerDidUpdateValueForCharacteristicEvent,
  BleState,
  BleStopScanEvent,
  Peripheral,
  PeripheralInfo,
} from 'react-native-ble-manager'

import { useDeviceStore } from '../stores/devices'
import { logger } from './logger'

type QueueType = {
  command: string
  response: string | null
  callback: (response: string | null, error?: any) => void
}

type ListenerType = {
  BleManagerStopScan: BleStopScanEvent
  BleManagerConnectPeripheral: BleConnectPeripheralEvent
  BleManagerDidUpdateState: BleManagerDidUpdateStateEvent
  BleManagerDiscoverPeripheral: BleDiscoverPeripheralEvent
  BleManagerDisconnectPeripheral: BleDisconnectPeripheralEvent
  BleManagerDidUpdateValueForCharacteristic: BleManagerDidUpdateValueForCharacteristicEvent
}

const addListener = <T extends keyof ListenerType>(
  eventType: T,
  listener: (event: ListenerType[T]) => void,
  context?: Object
) => BleManager.addListener(eventType, listener, context)

const removeListener = <T extends keyof ListenerType>(eventType: T) => {
  BleManager.removeAllListeners(eventType)
}

export type InitErrorType = 'Permissions' | 'Bluetooth' | 'Unknown'

const devicesState = useDeviceStore.getState()

class BleHelper {

  public readonly SERVICE_CHANNEL = '6e400001-b5a3-f393-e0a9-e50e24dcca9e'

  public readonly READ_CHANNEL = '6e400002-b5a3-f393-e0a9-e50e24dcca9e'

  public readonly WRITE_CHANNEL = '6e400003-b5a3-f393-e0a9-e50e24dcca9e'

  public readonly manager = BleManager

  public timerTimeout = 50
[... 304 lines cut ...]
    this.queue.push({ command, callback, response: null })
  }

  public eraseQueue() {
    this.queue = []
    this.breakQueueItem(null)
  }

  public destroy() {
    removeListener('BleManagerConnectPeripheral')
    removeListener('BleManagerDisconnectPeripheral')
    removeListener('BleManagerDiscoverPeripheral')
    removeListener('BleManagerStopScan')
    removeListener('BleManagerDidUpdateValueForCharacteristic')

    this.inited = false
  }
}

const bleHelper = new BleHelper()

export default bleHelper
```

Android: 12 SKQ1.211202.001

SaNFeeDep, 2024-11-13:
I found a link to GitHub in the application I use - https://github.com/kai-morich/SimpleBluetoothTerminal

marcosinigaglia, 2024-11-19:
Try to increase the MTU and use the maxByteSize parameter to send all the data in a single write.

SaNFeeDep, 2024-11-19:
@marcosinigaglia 
I tried increasing the MTU size and sending the response in maxByteSize, but it didn't work :(
My device does not support MTU greater than 23. I tried lowering the MTU size but it did not help.

SaNFeeDep, 2024-11-25:
I found a solution to the problem by talking with the firmware developer of the device.

The problem turned out to be that when sending the command it was necessary to add line breaks and carriage returns -> \n\r.

After adding these characters, the response from the device began to arrive instantly

closed 2024-11-25 by marcosinigaglia