innoveit/react-native-ble-manager #1291: [Android] "Error, gatt is null" during nextCommand processing leaves callback unfulfilled

polovi, 2024-11-26:
**Describe the bug**
"Error, gatt is null" during nextCommand processing leaves callback unfulfilled. Whole commandQueue is cleared. There is no response back to react-native that some error happen during processing.

https://github.com/innoveit/react-native-ble-manager/blob/7cbb6e5c63f1a274b8bf43c1901caa807feb3191/android/src/main/java/it/innove/Peripheral.java#L943-L948

**To Reproduce**
Hard to reproduce, we have a device that for some reason sometimes ends up in a strange state and breaks bluetooth communication. If this error can happen, it should be handled better.

**Expected behavior**
Errors should be handled inside commands and properly reported back through their callback. 

**Version:**
 - react-native-ble-manager version: 11.5.7

**Additional context**
```
2024-11-26 17:42:01.157 17752-17823 RNBleManager            com.rnapp                  D  Connect to: FB:83:98:81:E6:0D
2024-11-26 17:42:01.161 17752-17752 RNBleManager            com.rnapp                  I  DiscoverPeripheral: null
2024-11-26 17:42:01.213 17752-17752 RNBleManager            com.rnapp                  D  Is Or Greater than M $mBluetoothDevice
2024-11-26 17:42:01.238 17752-18853 RNBleManager            com.rnapp                  D  onConnectionStateChange to 2 on peripheral: FB:83:98:81:E6:0D with status 0
2024-11-26 17:42:01.249 17752-17752 RNBleManager            com.rnapp                  D  Peripheral event (BleManagerConnectPeripheral):FB:83:98:81:E6:0D
2024-11-26 17:42:01.249 17752-17752 RNBleManager            com.rnapp                  D  Connected to: FB:83:98:81:E6:0D
2024-11-26 17:42:01.453 17752-17765 RNBleManager            com.rnapp                  D  onConnectionStateChange to 0 on peripheral: FB:83:98:81:E6:0D with status 0
2024-11-26 17:42:01.504 17752-17752 RNBleManager            com.rnapp                  D  Peripheral event (BleManagerDisconnectPeripheral):FB:83:98:81:E6:0D
2024-11-26 17:42:02.596 17752-17823 RNBleManager            com.rnapp                  D  Retrieve services from: FB:83:98:81:E6:0D
2024-11-26 17:42:02.603 17752-17823 RNBleManager            com.rnapp                  D  Error, gatt is null
```

matt-42, 2024-12-10:
Same problem here, it seems that the commands are already handling the gatt == null errors but these lines prevent them to actually deal with the error. 
Theses lines were introduced 4 years ago by @dbarth to serialize the commands:
https://github.com/innoveit/react-native-ble-manager/commit/435ac0c6923a1577bd1fcf35f20ef514f049d702

Removing those two lines fixed the issue for me. I'm posting a PR, can anyone review it ?
https://github.com/innoveit/react-native-ble-manager/pull/1297

vhakulinen, 2025-02-03:
Much simpler approach would've been to just remove the null check from `nextCommand` and let the queued commands do their own checking when needed.

closed 2025-11-08 by marcosinigaglia