innoveit/react-native-ble-manager #1292: Need android.permission.BLUETOOTH_SCAN permission for AttributionSource

mMarcos208, 2024-11-27:
When Scan using android 14 I get this error

*AndroidManifest.xml*

    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE"/>

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28"/>

    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />

    
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
    

```
export const bluetoohPermission = async () => {
  if (Platform.OS === 'ios') {
    return true
  }
  if (Platform.OS === 'android' && Platform.Version >= 31) {
    const result = await PermissionsAndroid.requestMultiple([
      PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
      PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
    ])
    return !!result
  } else if (Platform.OS === 'android' && Platform.Version >= 23) {
    const result = await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION)

    if (result) {
      return true
    }
    const granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION)

    return granted === PermissionsAndroid.RESULTS.GRANTED
  }
}```

```
  const handleScan = async () => {
    try {
      if (!scanStatus) {
        setScanStatus('running')
        setDevices(new Map<BluetoothDevice['id'], BluetoothDevice>())

        const aceptPermission = await bluetoohPermission()

        if (!aceptPermission) {
          setScanStatus(undefined)
          notify('Need permissions.', 'warning')
          return
        }

        const isBluetoothEnabled = await RNBluetoothClassic.isBluetoothEnabled()

        if (!isBluetoothEnabled) {
          const userEnableBluetooh = await RNBluetoothClassic.requestBluetoothEnabled()

          if (!userEnableBluetooh) {
            setScanStatus(undefined)
            notify('You need enable bluetooth to start.', 'warning')
            return
          }
        }
        Geolocation.getCurrentPosition(
          (_position) => {
            getAvaliablePeripherals()
          },
          async (_error) => {
            setScanStatus(undefined)
            notify('Need location enable.', 'warning')
            if (Platform.OS === 'android') {
              await sleep(2000)
              Linking.sendIntent('android.settings.LOCATION_SOURCE_SETTINGS')
            }
          },
          {enableHighAccuracy: false, timeout: 5000, maximumAge: 10000},
        )
      }
    } catch (_) {
      setScanStatus(undefined)
      notify('ErroR', 'error')
    }
  }
  ```
  
  ```
const getAvaliablePeripherals = async () => {
    try {
      const avaliablePeripherals = await RNBluetoothClassic.startDiscovery()

      const addressPatern = /^([0-9A-Fa-f]{2}:){5}[0-9A-Fa-f]{2}$/

      avaliablePeripherals
        .filter((device) => !addressPatern.test(device.name))
        .forEach(({id, address, name}) =>
          addOrUpdateDevice(id, {id, address, name, bluetoothType: 'classic'}),
        )

      await BleManager.scan([], SECONDS_TO_SCAN_FOR, ALLOW_DUPLICATES, {
        matchMode: BleScanMatchMode.Sticky,
        scanMode: BleScanMode.LowLatency,
        callbackType: BleScanCallbackType.AllMatches,
      })
    } catch (error) {
      setScanStatus(undefined)
      notifyBugsnagWithSnapbar(error, 'Error.')
    }
  }```
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: motorola edge 40
 - OS: 14
 - react-native-ble-manager version: 11.5.5
 - react-native version: 0.76.3

marcosinigaglia, 2024-11-28:
The error is clear, you start the scan without the permission.

closed 2024-11-28 by marcosinigaglia

mMarcos208, 2024-11-28:
@marcosinigaglia 

I call  before Scan.

`const aceptPermission = await bluetoohPermission()`

stianjensen, 2024-11-28:
Check the return value from PermissionsAndroid.requestMultiple. It is not a bool, yet you seem to cast it into one, so in your code I believe it will always return true, even though permissions are rejected:
https://reactnative.dev/docs/0.73/permissionsandroid#requestmultiple

```javascript
const result = await PermissionsAndroid.requestMultiple([
  PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
  PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
])
return !!result
```