innoveit/react-native-ble-manager #1301: Cannot find symbol and method does not override or implement a method from a supertype @Override

pgiki, 2024-12-12:
**Describe the bug**
I am running react-native `0.76.4` and react-native-ble-manager `12.1.0` and I am getting `error: cannot find symbol` and `error: method does not override or implement a method from a supertype @Override` as show in the error trace below.

`
error Failed to install the app. Command failed with exit code 1: ./gradlew app:installDebug -PreactNativeDevServerPort=8081
Note: node_modules/expo-task-manager/android/src/main/java/expo/modules/taskManager/TaskService.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:43: error: cannot find symbol
class BleManager extends NativeBleManagerSpec { ^ symbol: class NativeBleManagerSpec
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:142: error: cannot find symbol emitOnDidUpdateState(map); ^ symbol: method emitOnDidUpdateState(WritableMap) location: class BleManager.MyBroadcastReceiver
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:187: error: cannot find symbol emitOnPeripheralDidBond(map); ^ symbol: method emitOnPeripheralDidBond(WritableMap) location: class BleManager.MyBroadcastReceiver
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:231: error: method does not override or implement a method from a supertype @Override ^
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:296: error: cannot find symbol if (getCurrentActivity() == null) ^ symbol: method getCurrentActivity() location: class BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:300: error: cannot find symbol getCurrentActivity().startActivityForResult(intentEnable, ENABLE_REQUEST); ^ symbol: method getCurrentActivity() location: class BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:369: error: cannot find symbol emitOnStopScan(map); ^ symbol: method emitOnStopScan(WritableMap) location: class BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:434: error: cannot find symbol peripheral.connect(callback, getCurrentActivity(), options); ^ symbol: method getCurrentActivity() location: class BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:686: error: cannot find symbol emitOnDidUpdateState(map); ^ symbol: method emitOnDidUpdateState(WritableMap) location: class BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:699: error: method does not override or implement a method from a supertype @Override ^
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:704: error: method does not override or implement a method from a supertype @Override ^
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:774: error: method does not override or implement a method from a supertype @Override ^
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:935: error: method does not override or implement a method from a supertype @Override ^
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:952: error: method does not override or implement a method from a supertype @Override ^
node_modules/react-native-ble-manager/android/src/main/java/it/innove/CompanionScanner.java:74: error: cannot find symbol bleManager.emitOnCompanionPeripheral(peripheral.asWritableMap()); ^ symbol: method emitOnCompanionPeripheral(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/CompanionScanner.java:79: error: cannot find symbol bleManager.emitOnCompanionPeripheral(null); ^ symbol: method emitOnCompanionPeripheral(<null>) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/CompanionScanner.java:91: error: cannot find symbol bleManager.emitOnCompanionPeripheral(peripheral != null ? peripheral.asWritableMap() : null); ^ symbol: method emitOnCompanionPeripheral(peripheral[...] null) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/CompanionScanner.java:148: error: cannot find symbol bleManager.emitOnCompanionFailure(map); ^ symbol: method emitOnCompanionFailure(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/CompanionScanner.java:169: error: cannot find symbol bleManager.emitOnCompanionFailure(map); ^ symbol: method emitOnCompanionFailure(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/Peripheral.java:103: error: cannot find symbol bleManager.emitOnConnectPeripheral(map); ^ symbol: method emitOnConnectPeripheral(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/Peripheral.java:113: error: cannot find symbol bleManager.emitOnDisconnectPeripheral(map); ^ symbol: method emitOnDisconnectPeripheral(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/Peripheral.java:447: error: cannot find symbol bleManager.emitOnDidUpdateValueForCharacteristic(map); ^ symbol: method emitOnDidUpdateValueForCharacteristic(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/DefaultScanManager.java:182: error: cannot find symbol bleManager.emitOnStopScan(map); ^ symbol: method emitOnStopScan(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/DefaultScanManager.java:219: error: cannot find symbol bleManager.emitOnDiscoverPeripheral(map); ^ symbol: method emitOnDiscoverPeripheral(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/DefaultScanManager.java:254: error: cannot find symbol bleManager.emitOnStopScan(map); ^ symbol: method emitOnStopScan(WritableMap) location: variable bleManager of type BleManager
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManagerPackage.java:23: error: incompatible types: BleManager cannot be converted to NativeModule modules.add(new BleManager(reactApplicationContext)); ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
26 errors FAILURE: Build failed with an exception. * What went wrong:
Execution failed for task ':react-native-ble-manager:compileDebugJavaWithJavac'.
Compilation failed; see the compiler error output for details. * Try:
Run with --info option to get more log output.
Run with --scan to get full insights. BUILD FAILED in 20m 58s.
`

**To Reproduce**
Steps to reproduce the behavior:
1. Update dependencies to  react-native `0.76.4` and react-native-ble-manager to `12.1.0`
2. Install dependencies and run `yarn android`
3. App is expected to build succesfully on Android.
4. Seen error error as shown above

**Expected behavior**
Compiling on android without an issue

**Smartphone (please complete the following information):**
 - Device: Android
 -  react-native-ble-manager version: `12.1.0`
 - react-native version: `0.76.4` 

**Additional context**
Before upgrading to `12.1.0` the app was compiling well. So far I have tried clearing gradle caches, deleting lock files and re-installing everything on node_modules. The same code is currently working fine on iOS.

marcosinigaglia, 2024-12-13:
Are you using the new architecture?

pgiki, 2024-12-14:
It worked fine when I switched to using the new architecture; however, due to a bug in `react-native-maps,` I am forced to disable it for now. When I disable the new architecture, `12.1.0` cannot compile anymore.

pgiki, 2024-12-14:
Closing this issue since the readme doc says only new architecture is supported for this version.

closed 2024-12-14 by pgiki