innoveit/react-native-ble-manager #1302: Is it work for depth level sensors

udaymamidisetti, 2024-12-17:
I am working on react native app that take the data from the depth level sensors. It is successully connecting but when i try to using this function BleManager.startNotification() to read data. It is not working

marcosinigaglia, 2024-12-17:
Why do you say the method does not work? Are you using `onDidUpdateValueForCharacteristic`?

adityaranjan006, 2025-02-04:
I'm not getting the complete data sent from firmware end. Why? only some chuck of data, but i'm able to get notified with the updated data.

dev-thissolution, 2025-03-03:
@adityaranjan006 : did you find the solution? As I am facing the same issue.

lucaswitch, 2025-03-16:
> I'm not getting the complete data sent from firmware end. Why? only some chuck of data, but i'm able to get notified with the updated data.

negotiate higher mtu, keep in mind that IOS and Android have diferente maximum mtu sizes.

closed 2025-06-21 by marcosinigaglia