innoveit/react-native-ble-manager #1303: Android: ble scan not working on version 12.1.0

andreanmol, 2024-12-18:
BLEManager.scan seems to stop working on new version. On previous version its working correctly
Current behavior is that it keeps scanning and the async promise is neither getting resolved nor rejected. So it's hard to debug what broke in this new version

marcosinigaglia, 2024-12-18:
Did you changed the event handling in the app?

andreanmol, 2024-12-18:
> Did you changed the event handling in the app?

My bad. It was a silly mistake by me i used event emitters like these
listeners = [
        bleManagerEmitter.addListener(
          'BleManagerDiscoverPeripheral',
          onBleManagerDiscoverPeripheral,
        ),
        bleManagerEmitter.addListener(
          'BleManagerStopScan',
          onBleManagerStopScan,
        ),
      ]; 
I fixed it by replacing them with these:
BleManager.onDiscoverPeripheral(onBleManagerDiscoverPeripheral);
BleManager.onStopScan(onBleManagerStopScan);

closed 2024-12-18 by marcosinigaglia