innoveit/react-native-ble-manager #1309: Example issues - expo\RN latest versions

iprimo, 2024-12-23:
Hi,

Thank you for your great work! I based my project on the example in the repo, and everything worked great. However, I had to upgrade Expo\RN due to some requirements, and after that, everything fell apart.

I tried to create a clean project with the latest versions of Expo\RN and react-native-ble-manager, but it didn't work. I've spent almost a week on this issue and haven't made any progress.

I would really appreciate it if you could provide an example of react-native-ble-manager working with the latest versions of Expo and React Native.

Thanks again!

package.json
`
{
  "name": "ABC",
  "main": "expo-router/entry",
  "version": "1.0.0",
  "scripts": {
    "start": "expo start",
    "android": "expo run:android",
    "ios": "expo run:ios",
    "web": "expo start --web",
    "test": "jest --watchAll"
  },
  "jest": {
    "preset": "jest-expo"
  },
  "dependencies": {
    "@expo/vector-icons": "^14.0.2",
    "@react-navigation/native": "^7.0.0",
    "convert-string": "^0.1.0",
    "expo": "~52.0.20",
    "expo-font": "~13.0.2",
    "expo-linking": "~7.0.3",
    "expo-router": "~4.0.14",
    "expo-splash-screen": "~0.29.18",
    "expo-status-bar": "~2.0.0",
    "expo-system-ui": "~4.0.6",
    "expo-web-browser": "~14.0.1",
    "react": "18.3.1",
    "react-dom": "18.3.1",
    "react-native": "0.76.5",
    "react-native-ble-manager": "^12.1.2",
    "react-native-reanimated": "~3.16.1",
    "react-native-safe-area-context": "4.12.0",
    "react-native-screens": "~4.1.0",
    "react-native-web": "~0.19.13"
  },
  "devDependencies": {
    "@babel/core": "^7.25.2",
    "@types/react": "~18.3.12",
    "jest": "^29.2.1",
    "jest-expo": "~52.0.2",
    "react-test-renderer": "18.3.1",
    "typescript": "~5.3.3"
  },
  "private": true
}

`

marcosinigaglia, 2024-12-23:
Hi @iprimo , the example is already using expo 52.

closed 2024-12-23 by iprimo

iprimo, 2024-12-23:
doh... wasted a few weeks on old code. Big thumbs up to you guys for your great work. tnx