innoveit/react-native-ble-manager #1310: Assistance with BLE-Manager Integration

iprimo, 2024-12-24:
Hi,

I am trying to use ble-manager in two different ways to get things working; however, both methods are failing. I would appreciate it if you could assist.

#1:
I’ve been trying to use the latest example in the repository, but I keep encountering errors related to plugins. I attempted to modify the plugin code from TypeScript to JavaScript to resolve the issue, but it’s becoming too complex, and I believe I’m missing something. I would appreciate any help you can provide. (Screenshot of the error attached.)

For example, I converted all imports to:
`const { AndroidConfig, ConfigPlugin } = require('expo/config-plugins');
const { withBLEAndroidManifest } = require('./withBLEAndroidManifest');
const { withBluetoothPermissions } = require('./withBluetoothPermissions');
`
[image]

#2:
I also tried building a new Expo/RN project from scratch using the navigation template, but I keep getting the following error:

**(NOBRIDGE) ERROR: Error: BleManagerModule not found [Component Stack]. (Screenshot attached.)**

[image]

I would greatly appreciate your assistance in resolving these issues.

Thanks!

iprimo, 2024-12-26:
PS: when I am using the project example, I only use the example folder and replace line 
` "react-native-ble-manager": "../"`
with latest version of the package eg:
 "react-native-ble-manager": "12.x.x",

Then, I copy the Plugin folder from Root of repot to the example folder and change the "src" folder inside plugin folder to  "build" folder.

marcosinigaglia, 2025-01-08:
Hi @iprimo , try run `npm run build:plugin` in the project root folder.

iprimo, 2025-01-10:
Hi @marcosinigaglia , 
Thanks for your reply. 
I am facing the following attached errors.

[image]
[image]

iprimo, 2025-01-10:
Previous screenshot were commands executed in Root folder of the project. The one below, commands executed in 'example' folder.
[image]

iprimo, 2025-01-30:
Found a way to fix the issue - Closing the case

closed 2025-01-30 by iprimo

cjswright, 2025-03-02:
Hi @iprimo - I'm having the exact same issue trying to use react-native-ble-manager within my own library with my app refusing to load it.

Could you elaborate on what you did to resolve?