innoveit/react-native-ble-manager #1326: RCTResponseSenderBlock Bridging Not Compiling Properly

dmyoung1994, 2025-01-27:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
Including the react-native-ble-manager in a project and building the project for iOS results in a compile bug where `RCTResponseSenderBlock` is not included properly

**To Reproduce**
Steps to reproduce the behavior:
1. Include react-native-ble-manger in expo project
2. Use the events that get called like `onDiscoverPeripheral`
3. Run `npx expo run:ios`
4. See error

**Expected behavior**
Including the library in a project compiles correctly.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6] iPhone 15 (physical device and simulator)
 - OS: [e.g. iOS8.1] iOS 17.6.1
 - react-native-ble-manager version: 12.1.3
 - react-native version: 0.76.6

**Additional context**
Full package.json dependencies
```
  "dependencies": {
    "@expo/config-plugins": "~9.0.0",
    "@expo/prebuild-config": "~8.0.0",
    "@expo/vector-icons": "^14.0.0",
    "@react-native-async-storage/async-storage": "1.23.1",
    "expo": "~52.0.0",
    "expo-blur": "~14.0.3",
    "expo-build-properties": "~0.13.2",
    "expo-camera": "~16.0.14",
    "expo-constants": "~17.0.4",
    "expo-contacts": "~14.0.3",
    "expo-device": "~7.0.2",
    "expo-linking": "~7.0.4",
    "expo-location": "~18.0.5",
    "expo-router": "~4.0.17",
    "expo-status-bar": "~2.0.1",
    "expo-system-ui": "~4.0.7",
    "react": "18.3.1",
    "react-native": "0.76.6",
    "react-native-ble-manager": "^12.1.3",
    "react-native-get-random-values": "~1.11.0",
    "react-native-permissions": "^5.1.0",
    "react-native-qrcode-svg": "^6.3.2",
    "react-native-safe-area-context": "4.12.0",
    "react-native-screens": "~4.4.0",
    "react-native-share": "^12.0.3",
    "react-native-svg": "15.8.0",
    "uuid": "^11.0.3"
  },
```

dmyoung1994, 2025-01-27:
After messing around and getting ever package updated properly, this no longer occurs

K-Jean, 2025-01-28:
@dmyoung1994 Hi ! I have the same error, do you know which package was the problem ?

CrazyGitter, 2025-02-07:
> [@dmyoung1994](https://github.com/dmyoung1994) Hi ! I have the same error, do you know which package was the problem ?

Hello, have you solved the problem？

CrazyGitter, 2025-02-07:
> After messing around and getting ever package updated properly, this no longer occurs

Hello, have you solved the problem？

dmyoung1994, 2025-02-07:
Sorry for the late reply! I was actually mistaken a was running off an
older version and not the newer one. Still having this issue.

faxioman, 2025-02-13:
Any feedback on this?

dmyoung1994, 2025-02-13:
No progress yet on my end. Not an expert at native libraries :/

faxioman, 2025-02-18:
Solved. I had to enable the new architecture in my Expo project, as clearly stated in the README.

lucaswitch, 2025-03-16:
I guess we can close this issue.
For anyone having this problem keep in mind to use ^12.x.x for new arch apps(rn >0.76), for old archs use the 11.x.x

EngsShi, 2025-03-21:
same issue

"react-native": "0.76.7",
"react-native-ble-manager": "^12.1.4",

[image]

lucaswitch, 2025-03-21:
> same issue
> 
> "react-native": "0.76.7", "react-native-ble-manager": "^12.1.4",
> 
> [image]

Make sure do enable new arch and refresh the binaries using: bundle exec pod install or something similar

closed 2025-06-27 by marcosinigaglia