innoveit/react-native-ble-manager #1330: Read and StartNotification MTU

adityaranjan006, 2025-02-04:
**Describe the bug**
For read() method and startNotification() method for the MTU I get is only 20 bytes, and there is no functionality to increase the MTU for read and onDidUpdateValueForCharacteristic() listerner for startNotification.

**Expected behavior**
I want to have a specific read and StartNotification MTU for onDidUpdateValueForCharacteristic() methods. By default its reading and notifying for 20 bytes only. Because I want it to be platform independent App( IOS and Android).

**Additional context**
Because of this multiple read requests were triggered on firmware BLE. Firmware is send the complete data but I'm not able to get it complete as its reading and notifying in fragmented manner.

marcosinigaglia, 2025-02-04:
https://innoveit.github.io/react-native-ble-manager/methods/#requestmtuperipheralid-mtu-android-only-api-21
If is not working is a device problem.

adityaranjan006, 2025-05-29:
what is the maximum MTU, for read? I was trying to to read I'm getting only 512 at a time in android? doesn't it reads automatically more all the data from the buffer? on IOS also 765 bytes?