innoveit/react-native-ble-manager #1332: docs: add table of contents for methods and events

jzaefferer, 2025-02-05:
The documentation for [methods](https://innoveit.github.io/react-native-ble-manager/methods/) and [events](https://innoveit.github.io/react-native-ble-manager/events/) would benefit from a table of contents (ToC), making it easier to scan the available items and jumping directly to a specific one. Especially the long list of methods would be easier to use.

If this sound interesting, I'd try to send a PR adding a ToC to https://github.com/innoveit/react-native-ble-manager/blob/master/docs/methods.markdown and https://github.com/innoveit/react-native-ble-manager/blob/master/docs/events.markdown

vhakulinen, 2025-02-07:
Even better would be to use something like [typedoc](https://typedoc.org/) to generate the documentation from the typescript files.

marcosinigaglia, 2025-02-07:
Hi @jzaefferer , your PR is welcome.

PR #1334 (2025-02-10): docs: add table of contents to events and methods
There's 35 methods, so that ToC starts closed.

Also adds instructions for testing docs locally in the README

Fixes #1332
closed 2025-02-10 by marcosinigaglia