innoveit/react-native-ble-manager #1335: Crash logs that seem to be caused by react-native-ble-manager. NullPointerException: Attempt to invoke virtual method 'void and Error receiving broadcast Intent

CaptainJeff, 2025-02-10:
**Describe the bug**
I'm only seeing this issue in the crashlogs. I'm unable to reproduce locally but it seems like its happening to a good number of people

     Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'void com.facebook.react.bridge.CxxCallbackImpl.invoke(java.lang.Object[])' on a null object reference
       at it.innove.NativeBleManagerSpec.emitOnDidUpdateState(NativeBleManagerSpec.java:46)
       at it.innove.BleManager$MyBroadcastReceiver.onReceive(BleManager.java:142)
       at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1698)
       at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(:2)
       at android.os.Handler.handleCallback(Handler.java:978)
       at android.os.Handler.dispatchMessage(Handler.java:104)
       at android.os.Looper.loopOnce(Looper.java:238)
       at android.os.Looper.loop(Looper.java:357)
       at android.app.ActivityThread.main(ActivityThread.java:8103)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1026)

          Fatal Exception: java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x5000010 (has extras) } in it.innove.BleManager$2@48d9436
       at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1708)
       at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(:2)
       at android.os.Handler.handleCallback(Handler.java:978)
       at android.os.Handler.dispatchMessage(Handler.java:104)
       at android.os.Looper.loopOnce(Looper.java:238)
       at android.os.Looper.loop(Looper.java:357)
       at android.app.ActivityThread.main(ActivityThread.java:8103)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1026)

**Smartphone (please complete the following information):**
 - Device: Moto G Pure (XT2163DL) [Android 12]
 - Device: Galaxy A13 5G [Android 13]

 - react-native-ble-manager version: 12.0.0
 - react-native version: 0.76.6

**Additional context**
I'm not totally sure what is happening because this isn't happening for everyone but i'm seeing a good number of crashes come through with these logs. Any help would be appreciated.

marcosinigaglia, 2025-02-11:
Hi @CaptainJeff , I thinks is a RN issue, can you try the RN 0.76.7?

CaptainJeff, 2025-02-11:
@marcosinigaglia yeah, I'll update to `76.7` but cause I can't reproduce it locally  won't know until I inspect the crash logs on production.

Any idea what commit between 76.6 => 76.7 would have fixed it?

CaptainJeff, 2025-02-11:
@marcosinigaglia ahh nevermind I think I found it. https://github.com/facebook/react-native/commit/5f1134c8df35efbc83163ca0b7d200f1fd6b4904

Although I'm not sure I totally understand `SurfaceMountingManager` seems like its responsible for managing the mounting and unmounting of React components on the native side and wasn't creating / destroying the event emitters properly?

marcosinigaglia, 2025-02-11:
I think so, but if you want some more information I think there was an issue.

closed 2025-02-11 by marcosinigaglia

CaptainJeff, 2025-02-13:
@marcosinigaglia this didn't seem to resolve the error I'm still getting htis with `76.7` 

          Fatal Exception: java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x5000010 (has extras) } in it.innove.BleManager$2@99860c6
       at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1708)
       at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(:2)
       at android.os.Handler.handleCallback(Handler.java:978)
       at android.os.Handler.dispatchMessage(Handler.java:104)
       at android.os.Looper.loopOnce(Looper.java:238)
       at android.os.Looper.loop(Looper.java:357)
       at android.app.ActivityThread.main(ActivityThread.java:8103)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1026)

          Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'void com.facebook.react.bridge.CxxCallbackImpl.invoke(java.lang.Object[])' on a null object reference
       at it.innove.NativeBleManagerSpec.emitOnDidUpdateState(NativeBleManagerSpec.java:46)
       at it.innove.BleManager$MyBroadcastReceiver.onReceive(BleManager.java:142)
       at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1698)
       at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(:2)
       at android.os.Handler.handleCallback(Handler.java:978)
       at android.os.Handler.dispatchMessage(Handler.java:104)
       at android.os.Looper.loopOnce(Looper.java:238)
       at android.os.Looper.loop(Looper.java:357)
       at android.app.ActivityThread.main(ActivityThread.java:8103)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1026)