innoveit/react-native-ble-manager #1336: App crashes after 4-5 seconds of calling startScan on signed apk - Android

barakataboujreich, 2025-02-18:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
on specific android devices, after calling BleManager.scan the app crashes after 4-5 seconds on signed APK
The error reported on firebase is not giving any details:
```
Crashed: Thread: SIGSEGV  0x00000000c0deadc4
#00 pc 0x326534 libart.so (BuildId: d963fb24d06855807a6574ea74a2cf53)
#01 pc 0xba09eda6
#02 pc 0xba09e544
#03 pc 0xba09ed86
#04 pc 0x322dfe libart.so (BuildId: d963fb24d06855807a6574ea74a2cf53)
#05 pc 0xba09e544
```

so i digged further with adb logcat and i noticed couple of things related to the crash:
```
2025-02-18	10:20:02.048	22972	22972	I		RNBleManager		DiscoverPeripheral: null
2025-02-18	10:20:02.057	2130	2397	D		BtGatt.GattService		onScanResult to scannerId: 3- eventType=0x1a, addressType=1, address=XX:XX:XX:XX:42:47, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-78, periodicAdvInt=0x0
2025-02-18	10:20:02.127	2130	2397	D		BtGatt.GattService		onScanResult to scannerId: 3- eventType=0x10, addressType=1, address=XX:XX:XX:XX:3E:98, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-58, periodicAdvInt=0x0
2025-02-18	10:20:02.264	22972	22972	F		libc		Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xc0deadc4 in tid 22972 (com.bletest), pid 22972 (com.bletest)
2025-02-18	10:20:02.340	23057	23057	I		crash_dump32		obtaining output fd from tombstoned, type: kDebuggerdTombstoneProto
2025-02-18	10:20:02.341	621	621	I		tombstoned		received crash request for pid 22972
2025-02-18	10:20:02.342	23057	23057	I		crash_dump32		performing dump of process 22972 (target tid = 22972)
2025-02-18	10:20:02.800	2130	2478	D		BtGatt.GattService		Binder is dead - unregistering scanner (3)!
2025-02-18	10:20:02.800	2130	2478	D		LeAppInfo		removeLeacReportedServerApp, appName: com.bletest
2025-02-18	10:20:02.801	802	802	I		Zygote		Process 22972 exited due to signal 11 (Segmentation fault)
```

i noticed that the crash is being dumped to Tombstone so i ran `adb.exe bugreport` to pull the tombstone file. and it contained
```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'samsung/a13vensser/a13ve:14/UP1A.231005.007/A137FXXU6EXG3:user/release-keys'
Revision: '1'
ABI: 'arm'
Processor: '1'
Timestamp: 2025-02-18 10:20:02.370105320+0000
Process uptime: 11s
Cmdline: com.bletest
pid: 22972, tid: 22972, name: com.bletest  >>> com.bletest <<<
uid: 10263
signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xc0deadc4
    r0  c0dead90  r1  e42ae809  r2  0000000d  r3  000000a5
    r4  fff61294  r5  fff61284  r6  fff61274  r7  0000206e
    r8  00000000  r9  e91baa00  r10 e1630e00  r11 ba09eda8
    ip  00000000  sp  fff61260  lr  e163454c  pc  e1634534

30 total frames
backtrace:
      #00 pc 00326534  /apex/com.android.art/lib/libart.so (nterp_op_invoke_virtual+52) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #01 pc 00876da8  /data/app/~~v5p35KBpWZ0nsdHyyvV0yQ==/com.bletest-7FoSQHRwBdIS8-dMoCMLFg==/oat/arm/base.vdex (it.innove.NativeBleManagerSpec.emitOnDiscoverPeripheral+16)
      #02 pc 0032c500  /apex/com.android.art/lib/libart.so (nterp_helper+2800) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #03 pc 00876546  /data/app/~~v5p35KBpWZ0nsdHyyvV0yQ==/com.bletest-7FoSQHRwBdIS8-dMoCMLFg==/oat/arm/base.vdex (it.innove.DefaultScanManager.onDiscoveredPeripheral+318)
      #04 pc 0032c500  /apex/com.android.art/lib/libart.so (nterp_helper+2800) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #05 pc 008763c4  /data/app/~~v5p35KBpWZ0nsdHyyvV0yQ==/com.bletest-7FoSQHRwBdIS8-dMoCMLFg==/oat/arm/base.vdex (it.innove.DefaultScanManager.-$$Nest$monDiscoveredPeripheral+0)
      #06 pc 0032ba48  /apex/com.android.art/lib/libart.so (nterp_helper+56) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #07 pc 00876234  /data/app/~~v5p35KBpWZ0nsdHyyvV0yQ==/com.bletest-7FoSQHRwBdIS8-dMoCMLFg==/oat/arm/base.vdex (it.innove.DefaultScanManager$2$1.run+12)
      #08 pc 005a2c87  /system/framework/arm/boot-framework.oat (android.os.Handler.dispatchMessage+70) (BuildId: 72d9f3b9c14393f5511d5220f35de6d6627160bb)
      #09 pc 005a5847  /system/framework/arm/boot-framework.oat (android.os.Looper.loopOnce+854) (BuildId: 72d9f3b9c14393f5511d5220f35de6d6627160bb)
      #10 pc 005a547f  /system/framework/arm/boot-framework.oat (android.os.Looper.loop+478) (BuildId: 72d9f3b9c14393f5511d5220f35de6d6627160bb)
      #11 pc 00383c9f  /system/framework/arm/boot-framework.oat (android.app.ActivityThread.main+1542) (BuildId: 72d9f3b9c14393f5511d5220f35de6d6627160bb)
      #12 pc 00143dd5  /apex/com.android.art/lib/libart.so (art_quick_invoke_stub_internal+68) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #13 pc 001bc1e1  /apex/com.android.art/lib/libart.so (void art::quick_invoke_reg_setup<true>(art::ArtMethod*, unsigned int*, unsigned int, art::Thread*, art::JValue*, char const*) (.__uniq.192663596067446536341070919852553954320.llvm.17112358095869631794)+112) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #14 pc 001bbd3f  /apex/com.android.art/lib/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+134) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #15 pc 0029e8f1  /apex/com.android.art/lib/libart.so (_jobject* art::InvokeMethod<(art::PointerSize)4>(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jobject*, _jobject*, unsigned int)+1100) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #16 pc 004c433f  /apex/com.android.art/lib/libart.so (art::Method_invoke(_JNIEnv*, _jobject*, _jobject*, _jobjectArray*) (.__uniq.165753521025965369065708152063621506277)+22) (BuildId: d963fb24d06855807a6574ea74a2cf53)
      #17 pc 0009c179  /system/framework/arm/boot.oat (art_jni_trampoline+56) (BuildId: 6baf4648bfc29351da5cc24a455a4e1c5e4a4b5d)
[... 16 lines cut ...]
[full logcat](https://github.com/user-attachments/files/18843481/logs_2025218_13206.txt)
[bugreport dumpstate-2025-02-18-10-20-35.zip](https://github.com/user-attachments/files/18843487/dumpstate-2025-02-18-10-20-35.zip) tombstone will be under: FS/data/tombstones/tombstone_01
[Git for test project that crashes](https://github.com/barakataboujreich/BleTestRn0.77)

**To Reproduce**
Steps to reproduce the behavior:
1- generate a release apk by `cd android && gradlew assembleRelease`
2- go to [Samsung Remote Test Lab](https://developer.samsung.com/remotetestlab/devices) sign in and get your free credits
3- search for device SM-A135F or SM-A137F and choose one from the list. it does not matter which android version or region, the crash happens on all devices in the list
4- upload the release apk to the device and install it
5- launch the app, you can wait for 40 seconds without doing anything and the app will crash on its own. or click Scan bluetooth and the app will crash after 5 seconds

Note: you can start a Remote Debug Bridge and launch the app with `yarn run android`, but it did not help me much since the app does not crash and behave as normal

**Expected behavior**
App will crash

**Smartphone (please complete the following information):**
 - Device: SM-A135F  SM-A137F (samsung a13)
 - OS: Android13 Android14
 - react-native-ble-manager version: 12.1.3
 - react-native version: 0.77.1 / 0.78.0-rc.5

**Additional context**
The crash that happens on it's own after 40 seconds of app launch and without doing anything is related to GC and can be checked in tombstone_00. i fixed it to be able to further debug the crash that is happening on scan by checking if mEventEmitterCallback is not null before calling any emit function in Spec file

h4shnerz, 2025-02-19:
Same issue, i assume happening on low end devices, weirdly it was fine on development phase

barakataboujreich, 2025-02-19:
> Same issue! Can't figure it out. Tried all sorts of patches.
> 
> I can't reproduce it but seeing the error logs in production. It does seem to be mostly related to Samsung A13 I've noticed

it started happening when we switched our production app to react native new architecture since we were forced to be able to publish on IOS. now we reverted back our android app to old react native version so users can continue using the app, we were receiving crash logs by the hundreds. 
At first i wasn't able to reproduce the crash on our testing devices, although not all were high end devices and we had phones similar to A13 specs, but the crash doesn't happen on them. 

so i decided to try one of the phones that are crashing, and the A13 was available on Samsung Remote test lab, it is a real device not emulator that you can test on and completely free you get 10 credits per day approx. 8 hours of testing.

if you test your app on it, it will crash

barakataboujreich, 2025-02-19:
Created an issue also on react-native facebook/react-native#49510

CaptainJeff, 2025-02-21:
@barakataboujreich any luck with this? I'm still struggling wtih this issue and seems to be happening to quite a few users. The weird thing is that it doesn't happen on debug - only release. Not sure why that would be the case

CaptainJeff, 2025-02-21:
@barakataboujreich i'm pretty sure the issue is the event emitters. 

For me at least, the error happens when the scan locates a device. The scan is fine but once there is a device returned from the scan then it crashes. When i comment out this line the crash stops https://github.com/innoveit/react-native-ble-manager/blob/67470d247a2cd2c7ca58e574b110d2fd7d2cc741/android/src/main/java/it/innove/DefaultScanManager.java#L224

Again - not happening on debug. Only release. Do you know if there is anything weird going on with these A13's in release related to event emitters?

CaptainJeff, 2025-02-22:
@barakataboujreich i might have found a solution. I haven't fully tested it yet but is working on samsung A13 (but won't get a chance to test on phones that currently work or ios yet)

Instead of sending events with  `bleManager.emitOnDiscoverPeripheral(map); `

```
reactContext
            .getJSModule(DeviceEventManagerModule.RCTDeviceEventEmitter.class)
            .emit("onDiscoverPeripheral", map);
```

And then in the app
```
const bleManagerEmitter = new NativeEventEmitter(NativeModules?.BleManager)

  useEffect(() => {
    const deviceListener = bleManagerEmitter.addListener(
      "onDiscoverPeripheral",
      handleDiscoverPeripheral
    )

    return () => {
      deviceListener?.remove()
    }
  }, [])
```

Here is a patch package to update all of the event emitter requests https://gist.github.com/CaptainJeff/b56113ee751658fe07f377b23e6a18ac

barakataboujreich, 2025-02-22:
@CaptainJeff yeah the issue is related to specific devices A13 is one of them. the list of devices we got crash on:
A13, A13 5G, A03 Core, A11, Moto G Pure, Lenovo Tab M8
at first i thought maybe it is a chip issue since the early reports were showing crashes on Mediatek chip only. but then crash reports kept on coming and the issue was happening on any chip. so i thought maybe it is a ram issue, but then some reports had 1.47GB+ free ram out of 4GB. so i went and bought one of the oldest cheapest Huawei phone that freezes if i open a browser, and surprisingly our app works normal and doesn't crash. so i have no idea what is the issue with these devices and i think neither does react-native team. 

i didn't try to solve the issue to be honest since it is coming from the event emitter that is auto generated by codegen. and our app is an access control system and we cannot afford users crashing and not being able to access buildings so i had to revert back to react native old architecture and wait for a solution.

thank you for the patch, i will try it on monday and test it on our live devices that works and report back to you

barakataboujreich, 2025-02-24:
@CaptainJeff i can confirm that the changes that you made fixed all the crashes that were happening. as i understood, you reverted back to the way this library used to send events from native code to JS before TurboModules. 
but i don't understand why the library changed the way it handles the event emission, maybe it was a requirement by react native when switching to TurboModules. 
i wish someone from the library contributors would have answered or appeared active on this topic...
Should i post on the react native issue thread, maybe someone can explain more about this crash and why it was solved by the change of methods? 
as much as i am happy about the fix, i am a little bit worried because if RN can't apply a proper fix, i will need to always make sure the libraries that we use and have event emitters will not cause the app to crash and maintain our version of the library code

barakataboujreich, 2025-02-25:
@marcosinigaglia thank you for the clarification. 
For now this is what i am doing, for android i have an old architecture project. but for IOS i had to have a new architecture project so i am able to publish the app to the store, and it is started to getting complicated managing two solutions and migrating every feature to the other solution. 

and i feel react native maintainers believe that the issue is with the library and not RN, and i do not know how to push them further more to investigate this issue since it is clear now that the cause of the crash is not the library itself but something related to new architecture. i opened an issue but it has gone inactive

marcosinigaglia, 2025-02-25:
The part of the code that handles events is very simple, it used to call the emitter or now passes through the code generated by codegen. 
If you can easily generate the exception try passing a default value to emitOnDiscoverPeripheral and see if that works. Perhaps it is something in the WritableMap that triggers the exception, from the stacktrace it is not clear.

barakataboujreich, 2025-02-25:
> The part of the code that handles events is very simple, it used to call the emitter or now passes through the code generated by codegen. If you can easily generate the exception try passing a [...]

ok i will try to debug it later on today. yeah the crash happens every time i start scanning on the A13 in samsung test lab, it never works. the mystery is that it only happens in release mode. if i run `yarn run android` the crash does not happen

barakataboujreich, 2025-03-04:
> Hi, I finally able to test using the samsung remote test lab. I used the example app and the latest expo without any issue. Probably something is wrong in your app and the new architecture is not set correctly.
> 
> [image]

is there any difference between expo and react-native-cli ?

also is this a release apk? because when i run it in debug mode `yarn run android` the app does not crash and behave normally

the crash only happens when i use a release apk `cd android && gradlew assembleRelease` then upload the apk to the device and install it, then i see the crash happens

marcosinigaglia, 2025-03-04:
It's a release with an apk. You can try the example project or try to use the same RN version and check if something is different. The expo result is not far from the react-native-cli but in my experience is more "solid".

barakataboujreich, 2025-03-04:
> It's a release with an apk. You can try the example project or try to use the same RN version and check if something is different. The expo result is not far from the react-native-cli but in my experience is more "solid".

i will need a bit of time to try to test expo/rn versions and see the issue from where it is generated. i am starting to feel it might be an environment setup issue.

barakataboujreich, 2025-03-07:
@marcosinigaglia i tried today a build with expo using development build, and generated the signed apk following the steps mentioned by [expo EAS](https://docs.expo.dev/build-reference/apk/) to remove any doubt of an environment setup issue. i used a custom template to apply your library, because i was not able to build the example app.

i had a different behavior the app was scanning but i was not able to see any devices in the list. i am 100% sure that i applied onDiscoverPeripheral the same way you apply it in the example app. And on my personal phone the same apk with the same android version but different phone model, the app scans and display the devices. I am not sure if the app did not crash because the event emitter were not called for some reason. the adb logcat shows the library scanning successfully on the SM-A135 

```
2025-03-07	12:54:46.164	2000	2169	D		BtGatt.GattService		onScanResult to scannerId: 4- eventType=0x1b, addressType=1, address=780E31_5, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-60, periodicAdvInt=0x0
2025-03-07	12:54:46.170	2879	2894	I		BluetoothAdapter		BluetoothAdapter() : com.bletest
2025-03-07	12:54:46.172	2000	2169	D		BtGatt.GattService		onScanResult to scannerId: 4- eventType=0x10, addressType=1, address=19A3F2_A, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-88, periodicAdvInt=0x0
2025-03-07	12:54:46.172	2879	2879	I		RNBleManager		DiscoverPeripheral: null
2025-03-07	12:54:46.183	2879	2879	I		RNBleManager		DiscoverPeripheral: null
2025-03-07	12:54:46.209	2000	2169	D		BtGatt.GattService		onScanResult to scannerId: 4- eventType=0x10, addressType=1, address=0E904F_4, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-89, periodicAdvInt=0x0
2025-03-07	12:54:46.210	2879	2879	I		RNBleManager		DiscoverPeripheral: null
2025-03-07	12:54:46.211	2000	2169	D		BtGatt.GattService		onScanResult to scannerId: 4- eventType=0x10, addressType=1, address=4612AC_7, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-71, periodicAdvInt=0x0
2025-03-07	12:54:46.218	2879	2879	I		RNBleManager		DiscoverPeripheral: null
2025-03-07	12:54:46.219	2000	2169	D		BtGatt.GattService		onScanResult to scannerId: 4- eventType=0x10, addressType=1, address=0851D2_0, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-56, periodicAdvInt=0x0
2025-03-07	12:54:46.228	2879	2879	I		RNBleManager		DiscoverPeripheral: null
```
question 1: did you try the expo app on the SM-135 or SM-137?

question 2: i am building the rn cli app on a windows machine with node version 22.12.0, java version "17.0.12". should i change any of these versions?

question 3: i cloned the library repo using the github desktop app. then went to the library directory and then `cd example && npm install`. then i tried `npx expo run:android`. and i tried on another clean clone just running `npx expo run:android` but i get the same error:
[image]

is this the correct way to run the example app? or am i missing something?

barakataboujreich, 2025-03-13:
@marcosinigaglia can you confirm please that you tried the signed apk on the SM-A135 or SM-A137?

marcosinigaglia, 2025-03-17:
> [@marcosinigaglia](https://github.com/marcosinigaglia) can you confirm please that you tried the signed apk on the SM-A135 or SM-A137?

Yes it was a SM-A135

CaptainJeff, 2025-03-17:
Might be worth mentioning that I wasn't getting this crash for my app when I ran it on samsung remote test lab but purchased an SM-A135 and it happened on that. 

As noted above this fixed the issue https://github.com/innoveit/react-native-ble-manager/issues/1336#issuecomment-2675886632 

But really not sure why event emitters would only have issues with these particular devices

barakataboujreich, 2025-03-17:
@CaptainJeff  that is weird, for my build i am having constant crashes on test lab. after your fix and a build from cli, i stopped facing crashes on testlab, so we tried a production release with the fix we started receiving too many crash reports. the app crashes instantly, although we do not start scanning or subscribe to any event on app launch, and the component that start scanning is not even loaded yet. only we initialize the library on app start in a useeffect.

on expo build the app does not crash, but i am not seeing any devices reported to JS. the adb logs shows the library is scanning, somehow the event is failing to report back but not crashing the app. i doubt that i am doing something wrong, because it is not that complicated to use this library as i used it perfectly fine before the new RN architecture.

there is too many weird behaviors that i had to revert back to old architecture again just so users are able to use the app...

@marcosinigaglia can you share some insights how to solve the build errors with the example app?

CaptainJeff, 2025-03-17:
@barakataboujreich really really strange.

Just to confirm. Before trying my fix was the app crashing instantly as well or was that introduced with my solution?

We have seen that same thing happen but i'm not sure if its related to this (it seemed to be happening prior to this patch but still on the new architecture). It seems to be happening on phones with really low memory available and seems to be a garbage collection error which I haven't been able to track down yet. But would be interesting if you only saw this happen once these changes were introduced.

I can't reproduce the instant crash even on my Galaxy a13 but do see users having the isseu based on crash logs

barakataboujreich, 2025-03-17:
> [@barakataboujreich](https://github.com/barakataboujreich) really really strange.
>
> Just to confirm. Before trying my fix was the app crashing instantly as well or was that introduced with my [...]

Sorry i might not have been clear with my previous statement. The crash related to GC was always there before/after you introduced your solution.

I just wanted to imply that on app start i don't even subscribe to any event emitter or start scanning and according to the log the app crashes within 2 seconds.

barakataboujreich, 2025-03-18:
@CaptainJeff 
if we make the changes you suggested in https://github.com/innoveit/react-native-ble-manager/issues/1336#issuecomment-2675886632
also remove from [NativeBleManager.ts](https://github.com/innoveit/react-native-ble-manager/blob/master/src/NativeBleManager.ts) line 3 and line 202 to 212, codegen will not generate any code in NativeBleManagerSpec referencing setEventEmitterCallback since we do not need them anymore, i suspect we will have a temporary working solution for all the crashes even GC one.

CaptainJeff, 2025-03-18:
@barakataboujreich 

Yeah, I probably should have removed those references from NativeBleManager.

What is your approach for the GC crash? Or are you saying you think my fix and removing those lines will fix it?

barakataboujreich, 2025-03-18:
@CaptainJeff 
yeah i suspect applying your fix and removing these lines will fix the GC crash, since the crash logs is pointing that GC destroyed something related to setEventEmitterCallback.
and when i applied both changes i event solved this crash [android 8](https://github.com/innoveit/react-native-ble-manager/issues/1319) without upgrading to RN 78 or applying changes suggested by [temp fix](https://github.com/facebook/react-native/issues/48009#issuecomment-2579944826)

CaptainJeff, 2025-03-18:
Yeah interesting. I'm still at a loss for whats causing the GC crash. But definitely could be related to this. I'll test it out

CaptainJeff, 2025-03-18:
@barakataboujreich damn. I didn't update ios to be compatible with this refactor yet. Have you tested on ios to see if the eventemitters done like this works?

barakataboujreich, 2025-03-18:
> [@barakataboujreich](https://github.com/barakataboujreich) damn. I didn't update ios to be compatible with this refactor yet. Have you tested on ios to see if the eventemitters done like this works?

no to be honest, i was too focused on android crashes and forgot about ios. but i think it will work since your fix is using the same event emitter that this library used to use before new architecture

CaptainJeff, 2025-03-18:
I actually ended up keeping ios the same and have two differerent hooks for each platform

useScaleListener.ios.ts
useScaleListener.android.ts

So i dont' have to update the objective-c / swift code. I haven't had a chance yet today to tackle this but was thinking i'll do the same thing for 

`NativeBleManager.ts`

and fork this repo and have 

NativeBleManager.ios.ts
NativeBleManager.android.ts

So i can just keep ios the same

CaptainJeff, 2025-03-18:
@barakataboujreich I just pushed the changes to a forked repo

https://github.com/coachcare/react-native-ble-manager

Going to test it out later today. Let me know if you are able to give it a test run or if i missed anything

CaptainJeff, 2025-03-18:
Awesome. Let me know. You can use this release https://github.com/coachcare/react-native-ble-manager/releases/tag/v.12.1.4-alpha

barakataboujreich, 2025-03-19:
i managed to install the repo from local folder, but not with git url. but codegen generated in NativeBleManagerSpec.java
```
protected final void emitOnDiscoverPeripheral(ReadableMap value) {
    mEventEmitterCallback.invoke("onDiscoverPeripheral", value);
  }

  protected final void emitOnStopScan(ReadableMap value) {
    mEventEmitterCallback.invoke("onStopScan", value);
  }

  protected final void emitOnDidUpdateState(ReadableMap value) {
    mEventEmitterCallback.invoke("onDidUpdateState", value);
  }
...
```
although in src i had NativeBleManager.android.ts and it didn't contain any relation to event emitter. maybe yarn is prioritizing NativeBleManager.ts ?

CaptainJeff, 2025-03-19:
@barakataboujreich. Hmm i don't think so but its possible. My guess is cause you are doing it locally you'll have to run `npm run build` to regenerate those directories - its probably using the old ones

If that doesn't work maybe change `NativeBleManager.ts` to `NativeBleManager.android.ts` or remove one of those values in `NativeBleManager.ts`

barakataboujreich, 2025-03-19:
yeah i did `npm run build` to build the library directories.

`NativeBleManager.ts` was taking priority over `NativeBleManager.android.ts`, i renamed it to `NativeBleManager.ios.ts` and the codegen did not generate any of the methods in Spec file. but now on ios it is not generating the files correctly, i had to rename it back from `NativeBleManager.ios.ts` to  `NativeBleManager.ts` for the files and to generate correctly and not have a build error. i do not mind now having two project one for android and one for ios and this minor change between them, as long as the crash on android are fixed. thanks

0x13b, 2025-03-30:
I confirm for **Xiaomi POCO C61** that the signed release application crashes silently in 5-10 seconds of idle run after app start or immediately after scan start. Debug version works fine and does not crash.

 **However,** both release and debug versions of the same app work fine on **Xiaomi Redmi Note 8 Pro**.

Above @h4shnerz [assumed](https://github.com/innoveit/react-native-ble-manager/issues/1336#issuecomment-2667420468) that this behavior could be related to low-end devices. As I know, POCO C61 operates on Android Go, so this might be the reason of such a weired behavior in release.

## Configuration
**Device:** Xiaomi POCO C61 (Android Go 14, API level 34) --> problem DOES exist
**Device:** Xiaomi Redmi Note 8 Pro (Android 10, API level 29) --> problem does NOT exist
**React Native:** v.0.78.1 (new architecture)
**react-native-ble-manager:** v12.1.4

Signed release app was configured in accord with [official docs](https://reactnative.dev/docs/signed-apk-android?package-manager=npm)

## Application
This is a single-button app that scans for devices when the button is pressed.
```js
import * as React from 'react';
import {StyleSheet, View, Button} from "react-native";
import getPermissions from "./permissions"
import {useEffect, useState} from "react";
import BleManager from "react-native-ble-manager"

function App() {

    useEffect(() => {
        getPermissions().then(console.log("Permissions requested."));
    }, [])

    useEffect(() => {
        const onStopListener = BleManager.onStopScan((args) => {
            setIsScanning(false);
            console.log("Scan stopped. ");
            console.log(args);
        });
        return () => {
            onStopListener.remove();
        };
    }, []);

    useEffect(() => {
        const onDiscoverListener = BleManager.onDiscoverPeripheral((args) => {
            console.log("Peripheral discovered.");
            console.log(args);
        });
        return () => {
            onDiscoverListener.remove();
        };
    }, []);

    const [isScanning, setIsScanning] = useState(false);

    const scanNow = async () => {
        await BleManager.start({ forceLegacy: false });
        console.log("BleManager initialized");
        setIsScanning(true);
        await BleManager.scan([], 10);
    }

    return (
        <View style={styles.wrapper}>
            <Button
                title={"Start scanning"}
                disabled={isScanning}
                onPress={() => scanNow()}
            />
        </View>
    );
}

[... 47 lines cut ...]
03-30 21:42:26.449  7541  7541 F DEBUG   :  field_offset=8 name=mEventEmitterCallback reference addr adjacent_ram=00370036 00390038 00420041 00440043 00460045 00000000 12ed4eb8 00000000 |f3fe1e09 12e8cb10 00000000 00000000 00000000
12ed5b90 12e8cb10 00000000 0xf3ffa180 Bump pointer space 1 0x12c00000-0x13115000 - 0x1ac00000
<~40 more lines> 
```
### Crash immediately after scan start
```
03-30 21:46:48.314  7330  7330 F libc    : Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xc0deadc4 in tid 7330 (com.ble), pid 7330 (com.ble)
03-30 21:46:48.938  7633  7633 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
03-30 21:46:48.938  7633  7633 F DEBUG   : Build fingerprint: 'POCO/blue_p_ru/blue:14/UP1A.231005.007/V816.0.10.0.UGRRUXM:user/release-keys'
03-30 21:46:48.938  7633  7633 F DEBUG   : Revision: '0'
03-30 21:46:48.938  7633  7633 F DEBUG   : ABI: 'arm'
03-30 21:46:48.938  7633  7633 F DEBUG   : Timestamp: 2025-03-30 21:46:48.467472409+0300
03-30 21:46:48.938  7633  7633 F DEBUG   : Process uptime: 406s
03-30 21:46:48.938  7633  7633 F DEBUG   : Cmdline: com.ble
03-30 21:46:48.938  7633  7633 F DEBUG   : pid: 7330, tid: 7330, name: com.ble  >>> com.ble <<<
03-30 21:46:48.938  7633  7633 F DEBUG   : uid: 10224
03-30 21:46:48.938  7633  7633 F DEBUG   : signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xc0deadc4
03-30 21:46:48.938  7633  7633 F DEBUG   :     r0  c0dead90  r1  f3fe1e09  r2  0000000d  r3  00007cf2
03-30 21:46:48.938  7633  7633 F DEBUG   :     r4  ffb82ec4  r5  ffb82eb4  r6  ffb82ea4  r7  0000206e
03-30 21:46:48.938  7633  7633 F DEBUG   :     r8  00000000  r9  f409dc00  r10 ee069100  r11 b65c06dc
03-30 21:46:48.938  7633  7633 F DEBUG   :     ip  00000000  sp  ffb82e90  lr  ee06c84c  pc  ee06c834
03-30 21:46:48.938  7633  7633 F DEBUG   : 30 total frames
03-30 21:46:48.938  7633  7633 F DEBUG   : backtrace:
<~40 more lines> 
```

barakataboujreich, 2025-03-31:
i do not think anymore that this is a problem with this library more than a problem with RN. i tried the example app provided by the library. i cloned the repo. `npm install` in root folder. `npm build` in root folder. `cd example` then `npm install`. then `cd android && .\gradlew assembleRelease`, i am still facing the same issue.

then i decided to drop the library and tried a simple native module with event emitting capabilities by following this guide [turbo modules](https://github.com/reactwg/react-native-new-architecture/blob/main/docs/turbo-modules.md)
whenever i include the eventemitter library to the codespec and codegen generates the code, the app starts crashing, there is something causing the issue, i don't know what, if java version, node version or npm version. 

i also opened an issue at react native repo, it is linked here, but i get the feeling from them that they think it is a library issue, and i am not able to convince them otherwise since i am the only one writing about that issue there. i don't know if it will help if we get active on that issue linked. i wrote there a complete detail of my finding but still no reply.

@marcosinigaglia can you share the node version, npm version and java version you used to build the release apk?

0x13b, 2025-03-31:
@barakataboujreich  I saw your issue and the corresponding discussion there. I'm not very good at mobile development, thus I didn't try anything more than `react-native-ble-manager` to reproduce the issue.

My node version is v23.9.0 (openjdk v17.0.2). I added this info to the post above

barakataboujreich, 2025-04-01:
@0x13b yes i saw you included the version details. i was asking the library maintainer about his versions since he built a release apk and the app behaved normal and didn't crash with his test.

i do not have the POCO devices to test. the A52 does not crash and our app behave normally. 
the list of devices that we received crash reports on them: A13, A13 5G, A03 Core, A11, Moto G Pure, Lenovo Tab M8

while i also do not have a physical device to test on those, i was able to use Samsung test lab a free platform to test on real samsung devices, unfortunately i only found the A13 on it which operates on normal android version not android GO...

if you are interested in testing on A13 go to [Remote test lab](https://developer.samsung.com/remotetestlab/devices) sign in you can use your google account if you do not have a samsung account. in the list of devices search for SM-A135N-KR1 this is the device i use and our app crashes, the test app crashes

0x13b, 2025-04-01:
@barakataboujreich Yeap, I've just tested my release apk at Samsung Test Lab on A13 (SM-A135N-KR3). It fails in a similar way as on my Xiaomi POCO C61 with same error log

```
F		com.ble		verification.cc:107] GC tried to mark invalid reference 0xea151609
2025-04-02	00:29:13.173	7233	7241	F		com.ble		verification.cc:107] ref=0xea151609 <invalid address>
2025-04-02	00:29:13.173	7233	7241	F		com.ble		verification.cc:107] holder=0x13462fc0 klass=0x13557be8(java.lang.Class<it.innove.BleManager>) space=main space (region space) 0x12c00000-0x2ac00000 card=0 adjacent_ram=132fd790 00000002 1300bc70 10000000 00000000 13002e78 132fead0 00000000 |13557be8 10000000 ea151609 13002e78 00000000 00000000 00000000 13019e30
2025-04-02	00:29:13.173	7233	7241	F		com.ble		verification.cc:107]  field_offset=8 name=mEventEmitterCallback reference addr adjacent_ram=1300bc70 10000000 00000000 13002e78 132fead0 00000000 13557be8 10000000 |ea151609 13002e78 00000000 00000000 00000000 13019e30 13002e78 00000000 0xed79c400 main space (region space) 0x12c00000-0x2ac00000
2025-04-02	00:29:13.173	7233	7241	F		com.ble		verification.cc:107] 0xed79c5a0 region space live bitmap[begin=0x12c00000,end=0x2ac00000]
2025-04-02	00:29:13.173	7233	7241	F		com.ble		verification.cc:107] 0xed79c5a0 region space live bitmap[begin=0x12c00000,end=0x2ac00000]
```

barakataboujreich, 2025-04-01:
@0x13b ok perfect at least we are consistent and we do not have any difference in the behavior. 
now for the bad news, i do not know if you are also following up on the other issue opened in React native repo
> Created an issue also on react-native [facebook/react-native#49510](https://github.com/facebook/react-native/issues/49510)

apparently they will not be able to follow up on this issue unless we are able to tell them how to replicate the issue on an emulator or a pixel device, and i was not able to replicate it so far on an emulator

0x13b, 2025-04-02:
@barakataboujreich Anyway I left a [comment](https://github.com/facebook/react-native/issues/49510#issuecomment-2772574324) in the issue you cited

marcosinigaglia, 2025-08-11:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

SCJMENGMENG, 2025-09-08:
"react-native-ble-manager": "12.1.5"，"react-native": "0.76.9"