innoveit/react-native-ble-manager #1339: ⚠️ BLE Notify Event Overwriting Previous Data Issue 🛑

jwAndroid, 2025-02-27:
**Describe the bug**
The issue occurs when the BLE notify event triggers multiple times with the same data. Instead of appending new data to the buffer, the new data overwrites the previous data, causing loss of information. This issue is causing incorrect processing and data inconsistency.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a BLE device.
2. Receive multiple notify events with the same data.
3. Observe that the previous data is overwritten with the new data, causing data loss.
4. Please check the screenshots and logs for more details.

**Expected behavior**
I expected that each new BLE notify event would append data to the buffer without overwriting the previous data. This would ensure that all received data is preserved correctly for further processing.

**Screenshots**
[image]
[image]
[image]
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: Galxy Note20 5G (SM-N981N)
 - OS: Android 13
 - react-native-ble-manager version: ^11.5.3
 - react-native version: 0.74.1

**BLE related code**
Related Code:
Here’s the source code related to the BLE notify handling:

1. HomeScreen
```javascript
const HomeScrn = () => {
  const data = useAppSelector((state) => state.mainSlice.getMain.response);
  const { appId, fcmToken } = useAppSelector((state) => state.userSlice.user);
  const { peripheralId } = useAppSelector((state) => state.bleConnectionSlice.connect.success);
  const { currentRoute } = useAppSelector((state) => state.viewSlice.navigation);
  const { validAuthHistCnt } = useAppSelector((state) => state.mainSlice.getMain.response);
  const { isTimerRunning } = useAppSelector((state) => state.viewSlice.globalTimmer);

  const { t } = useTranslation();

  const dispatch = useAppDispatch();

  const { navigate, pop } = useNavigation<RouterModel.NavigationProp>();

  const {
    onPressAuthHistory,
    onPressManageAuthInfo,
    onPressNoticeList,
    onPressQRCodeAuth,
    onPressSettings,
  } = useNavigate();

  const { abortShowModal } = useAuthService();

  ...

  useBle(); // here 
  
  ... 

  return (
    <SafeAreaContainer>
      <HomeHeader
        notificationCount={validAuthHistCnt}
        onPressFirst={onPressSettings}
        onPressSecond={() => {
          navigate('AuthRouter', { screen: 'AuthPending' });
          dispatch(getValidAuthHistReset());
[... 357 lines cut ...]
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 6abeb98dcfee4bb11c43812c62b44e56d5da4129ffa5a00cd51e7acbe18afdc2
 LOG  hash 555bd2e75bbc8de0e0d44a4389c405df988cf66231106e64dc2f69661987d035
```

**Additional context**
1. data diff open link
 - https://www.diffchecker.com/bKQ2QGMd/

**Additional**
useBle is used with Redux and declared globally to be used only once. The key point is that the value data provided in the handleUpdateValueForCharacteristic function should always be the same, but it changes unexpectedly. The data in message3 from the link to the diff is the original data provided in the value. message is the processed version of message3, and authDataHash is the result of hashing the stream.

The most critical issue is that message3 gets overwritten by the previous element, which is the main problem.

jwAndroid, 2025-02-27:
If you need any additional information, please feel free to ask.

jwAndroid, 2025-02-27:
@marcosinigaglia help me plz

marcosinigaglia, 2025-02-27:
Hi, I don't understand the problem, are you using `startNotificationWithBuffer`?

jwAndroid, 2025-02-28:
> Hi, I don't understand the problem, are you using `startNotificationWithBuffer`?

Hello! Thank you so much for your response.  
I do not use `startNotificationWithBuffer`.  
When connecting, I only use `startNotification`.  
Additionally, would you like me to share the entire source code related to the connection process?

jwAndroid, 2025-02-28:
Just in case, I will leave my code below.

```typescript
function* connectSaga(action: PayloadAction<BleConnectionModel.ConnectRequest>): SagaIterator {
  try {
    const { peripheralId } = action.payload;

    const { peripheralId: connectedId }: BleConnectionModel.ConnectSuccess = yield select(
      (state: RootState) => state.bleConnectionSlice.connect.success,
    );

    if (connectedId) {
      yield call(disconnect, connectedId);
    } else {
      yield call(stopScan);
    }

    yield delay(700);

    yield call(connect, peripheralId);

    // if (Platform.OS === 'android') {
    //   yield delay(700);
    //   yield call(requestMTU, peripheralId);
    // }

    const { characteristics, name } = yield call(retrieveServices, peripheralId);

    const writeItem = characteristics.find(
      (item: any) => item.properties.Write || item.properties.WriteWithoutResponse,
    );
    const notifyItem = characteristics.find((item: any) => item.properties.Notify);

    if (!notifyItem || !writeItem) {
      throw new Error('Notify, Write Error');
    }

    const peripheralObj = {
      notify: {
        serviceUUID: notifyItem.service,
        characteristicId: notifyItem.characteristic,
      },
      write: {
        serviceUUID: writeItem.service,
        characteristicId: writeItem.characteristic,
      },
    };

    yield delay(700);

    yield call(startNotification, {
      peripheralId,
      serviceUUID: peripheralObj.notify.serviceUUID,
      characteristicId: peripheralObj.notify.characteristicId,
    });

    yield delay(700);

    yield call(write, {
      peripheralId,
      serviceUUID: peripheralObj.write.serviceUUID,
      characteristicId: peripheralObj.write.characteristicId,
      command: bootRequestCommand(peripheralId),
    });

    yield put(connectSuccess({ peripheralId, name, write: peripheralObj.write }));

    yield delay(700);

    yield call(write, {
      peripheralId,
      serviceUUID: peripheralObj.write.serviceUUID,
      characteristicId: peripheralObj.write.characteristicId,
      command: versRequestCommand(),
    });
  } catch (error: any) {
    yield put(connectFailure(String(error)));
  }
}
```

marcosinigaglia, 2025-03-03:
I don't think it is a Scrcpy problem but I don't see why the problem should be in the library. You have to implement a protocol to avoid duplicates which may be due to a problem with the ble or device stack.

closed 2025-03-03 by marcosinigaglia