innoveit/react-native-ble-manager #1340: Error reading from BLE Error reading XXX-XXX status=2

Maxime149, 2025-02-27:
Hello everyone, 

I'm trying to communicate from a React Native app with a ble emulate device (With NrF Connect).

**Describe the bug**
When i'm trying to read a characteristic that i'm sure it's exist i have this error : Error Reading from BLE Error reading 01234567-0123-0123-0123-012345678912 status=2

I retrieve without any problems info of the device so I think I'm well connected.
I didn't succed to find the meaning of the status=2 error, does someone knows where we can find this error table.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect device : 

```ruby
export async function scanDevices() {
    BleManager.scan([], 10, false).then(() => {
        // Success code
        console.log("Scan started");
    });
    await timeout(10000);
    deviceId = await BleManager.getDiscoveredPeripherals().then(async (peripheralsArray) => {
        // Success code           
        console.log("Discovered peripherals: " + peripheralsArray.length);
        for (var i = 0; i < peripheralsArray.length; i++) {
            console.log(peripheralsArray[i])
        }
        testDevice = peripheralsArray.find(device => device.name === 'TestDevice');
        if (testDevice === undefined) {
            Alert.alert(
                'Failed :(',
                'Please launch again the scan',
                [{ text: 'OK' }]
            );
        }
        deviceId = testDevice ? testDevice.id : ""
        console.log("=====================");
        console.log("device: ")
        console.log(testDevice);
        console.log("id: " + deviceId);
        return deviceId
    });

    BleManager.connect(deviceId)
        .then(() => {
            // Success code
            console.log("Connected to " + testDevice);
            BleManager.retrieveServices(deviceId).then(
                (peripheralInfo) => {
                    // Success code
                    console.log("Peripheral info:", peripheralInfo);
                }
            );
            Alert.alert(
                'Connected !',
                'You are now connected to:' + testDevice?.name,
                [{ text: 'OK' }]
            );
        })
        .catch((error) => {
            // Failure code
            console.log(error);
        });
}
```

2. Try to read a characteristic
```ruby
export function readFromBle() {
    BleManager.retrieveServices(deviceId).then((response) => {
        BleManager.read(deviceId, serviceUUID, charcUUID)
            .then((resp) => {
                console.log("Read from BLE", resp)
            })
            .catch((err) => {
                console.log("=====================");
                console.error("Error Reading from BLE", err)
                console.log("id :" + deviceId)
                console.log("service UUID :" + serviceUUID)
                console.log("charac UUID :" + charcUUID)
                console.log(err);
                console.log("=====================");
            })
    })
}
```

**Expected behavior**
Have a return with the value of the characteristic

**Screenshots**

[image]

Here you can see that my characteristic exists
[image]

**Smartphone:**
 - Device: Xiaomi Redmi 12C
 - OS: MIUI Global 14.0.3 (Android 13)
 - react-native-ble-manager version: 12.1.4
 - react-native version: 0.78.0

marcosinigaglia, 2025-02-27:
Are you sure the characteristic is readable?

Maxime149, 2025-02-27:
To prove you the characteristic was readable I went in the parameter of my emulated BLE device (NRF Connect) I figure out there were properties AND permissions for write and read, only properties were activated.

I feel stupid, sorry for bothered you and thanks for your time

closed 2025-02-27 by Maxime149