innoveit/react-native-ble-manager #1344: unable to get battery percentage of device

prasanna-colan, 2025-03-20:


lucaswitch, 2025-03-20:
For battery mostly all devices uses the default battery protocol(see https://www.bluetooth.com/specifications/specs/bas-1-1/).
The common pattern to work with it:

- Verify if battery service and characteristic are present in the device.
- Read the characteristic value and then enable notification.

I'll quickly share what i have currently on production for anyone that don't know how to begin working with it:

## Getting the battery status 
```ts
// bytes its the actual characteristic value
export function getBluetoothBatteryLevelPackage(
  bytes: number[],
): BluetoothBatteryLevelPackage {
  let level = 0;
  if (bytes.length > 0) {
    level = bytes[0];
  }

  return new BluetoothBatteryLevelPackage(level, Date.now(), bytes); // or just return level
}
```

## Getting the charge status
Keep in mind that many devices does not have that kind of information available

// bytes its the actual characteristic value
```ts
export function getBluetoothBatteryLevelStatusPackage(
  bytes: number[],
): BluetoothBatteryLevelStatusPackage {
  const flagsField = bytes[0];
  const powerStatePresent = Boolean(flagsField & 0x01);
  const batteryLevelPresent = Boolean(flagsField & 0x04);

  let powerStatus = {
    battery_present: false,
    wired_external_power: BluetoothBatteryLevelStatusExternalPower.Unknown,
    wireless_external_power: BluetoothBatteryLevelStatusExternalPower.Unknown,
    battery_charge_state: BluetoothBatteryLevelStatusChargeState.Unknown,
    battery_charge_level: BluetoothBatteryLevelStatusChargeLevel.Unknown,
    charging_type: BluetoothBatteryLevelStatusChargeType.UnknownOrNotCharging,
  };

  if (powerStatePresent) {
    const powerByte = bytes[1];

    powerStatus.battery_present = Boolean(powerByte & 0x01);

    const wiredBits = (powerByte >> 1) & 0x03; // mask 2 bits
    switch (wiredBits) {
      case 0b00:
        powerStatus.wired_external_power =
          BluetoothBatteryLevelStatusExternalPower.No;
        break;
      case 0b01:
        powerStatus.wired_external_power =
          BluetoothBatteryLevelStatusExternalPower.Yes;
        break;
      case 0b10:
        powerStatus.wired_external_power =
          BluetoothBatteryLevelStatusExternalPower.Unknown;
        break;
    }

    // bits 3-4
    const wirelessBits = (powerByte >> 3) & 0x03; // mask 2 bits
    switch (wirelessBits) {
      case 0b00:
        powerStatus.wireless_external_power =
          BluetoothBatteryLevelStatusExternalPower.No;
        break;
      case 0b01:
        powerStatus.wireless_external_power =
          BluetoothBatteryLevelStatusExternalPower.Yes;
        break;
      case 0b10:
        powerStatus.wireless_external_power =
          BluetoothBatteryLevelStatusExternalPower.Unknown;
        break;
    }

    // bits 5-6
    const chargeStateBits = (powerByte >> 5) & 0x03; // mask 2 bits
    switch (chargeStateBits) {
      case 0b00:
        powerStatus.battery_charge_state =
          BluetoothBatteryLevelStatusChargeState.Unknown;
        break;
      case 0b01:
        powerStatus.battery_charge_state =
          BluetoothBatteryLevelStatusChargeState.Charging;
        break;
      case 0b10:
        powerStatus.battery_charge_state =
          BluetoothBatteryLevelStatusChargeState.DischargingActive;
        break;
      case 0b11:
        powerStatus.battery_charge_state =
          BluetoothBatteryLevelStatusChargeState.DischargingInactive;
        break;
    }

    if (batteryLevelPresent && bytes.length > 1) {
      const levelByte = bytes[1];
      const levelBits = levelByte & 0x03;
      switch (levelBits) {
        case 0b00:
          powerStatus.battery_charge_level =
            BluetoothBatteryLevelStatusChargeLevel.Unknown;
          break;
        case 0b01:
          powerStatus.battery_charge_level =
            BluetoothBatteryLevelStatusChargeLevel.Low;
          break;
        case 0b10:
          powerStatus.battery_charge_level =
            BluetoothBatteryLevelStatusChargeLevel.Critical;
          break;
        case 0b11:
          powerStatus.battery_charge_level =
            BluetoothBatteryLevelStatusChargeLevel.Good;
          break;
      }
    }

    // This is optional
    if (bytes.length > 2) {
      const chargingByte = bytes[2];
      // Assume Charging Type is in the lower 2 bits
      const chargingTypeBits = chargingByte & 0x03;
      switch (chargingTypeBits) {
        case 0b00:
          powerStatus.charging_type =
            BluetoothBatteryLevelStatusChargeType.UnknownOrNotCharging;
          break;
        case 0b01:
          powerStatus.charging_type =
            BluetoothBatteryLevelStatusChargeType.ConstantCurrent;
          break;
        case 0b10:
          powerStatus.charging_type =
            BluetoothBatteryLevelStatusChargeType.ConstantVoltage;
          break;
        case 0b11:
          powerStatus.charging_type =
            BluetoothBatteryLevelStatusChargeType.Trickle;
          break;
      }
    }
  }

  return new BluetoothBatteryLevelStatusPackage(
    batteryLevelPresent,
    powerStatus.wired_external_power,
    powerStatus.wireless_external_power,
    powerStatus.battery_charge_state,
    powerStatus.battery_charge_level,
    powerStatus.charging_type,
    Date.now(),
    bytes,
  );
}

// further types
export enum BluetoothBatteryLevelStatusExternalPower {
  Unknown = -1,
  Yes = 1,
  No = 0,
}

export enum BluetoothBatteryLevelStatusChargeState {
  Unknown = -1,
  Charging = 1,
  DischargingActive = 2,
  DischargingInactive = 3,
}

export enum BluetoothBatteryLevelStatusChargeLevel {
  Unknown = -1,
  Critical = 1,
  Low = 2,
  Good = 3,
}

export enum BluetoothBatteryLevelStatusChargeType {
  UnknownOrNotCharging = -1,
  ConstantCurrent = 1,
  ConstantVoltage = 2,
  Trickle = 3,
}

```

Also i can confirm this is not a library issue.

prasanna-colan, 2025-03-24:
{
  "advertising": {
    "manufacturerData": {
      "64b6": {
        "bytes": [
          104,
          183,
          210,
          189,
          246,
          218,
          179,
          218
        ],
        "data": "aLfSvfbas9o=",
        "CDVType": "ArrayBuffer"
      }
    },
    "txPowerLevel": -2147483648,
    "isConnectable": true,
    "serviceData": {
      
    },
    "localName": "FSRKB_BT-001",
    "serviceUUIDs": [
      "af30"
    ],
    "manufacturerRawData": {
      "bytes": [
        0,
        0,
        100,
        182,
        104,
        183,
        210,
        189,
        246,
        218,
        179,
        218
      ],
      "data": "AABktmi30r322rPa",
      "CDVType": "ArrayBuffer"
    },
    "rawData": {
      "bytes": [
        2,
        1,
        6,
        3,
        3,
        48,
        175,
        11,
        255,
        182,
        100,
        104,
        183,
        210,
        189,
        246,
        218,
        179,
        218,
        13,
        9,
        70,
        83,
        82,
        75,
        66,
        95,
        66,
        84,
        45,
        48,
        48,
        49,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0
      ],
      "data": "AgEGAwMwrwv/tmRot9K99tqz2g0JRlNSS0JfQlQtMDAxAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=",
      "CDVType": "ArrayBuffer"
    }
  },
  "rssi": -56,
  "id": "D2:BD:F6:DA:B3:DA",
  "name": "FSRKB_BT-001"
}
@lucaswitch  @typester  @seigel  @quatrix 
this is my response for my bluetooth connected device how i want to get my charge percentage

prasanna-colan, 2025-03-24:
@lucaswitch battery service not present in my response any otherways has to get battery percentage

lucaswitch, 2025-03-24:
> [@lucaswitch](https://github.com/lucaswitch) battery service not present in my response any otherways has to get battery percentage

This is peripheral info. There is no device that has the battery info inside the peripheral object, you need to read the actual battery service UUID + characteristic UUID.

closed 2025-06-30 by marcosinigaglia