innoveit/react-native-ble-manager #1347: EXC_BAD_ACCESS crash

thompsonnoah, 2025-03-21:
**Describe the bug**
After installing `react-native-ble-manager` in a new project, I get the following crash in XCode on the simulator as well as on the device.

`EXC_BAD_ACCESS: A bad access to memory terminated the process.`

The crash occurs here:

[image]

After installing and adding this code to `App.tsx`

```
useEffect(() => {
    BleManager.start({showAlert: true}).then(() => console.log('BleManager init'));
  }, []);
```

**To Reproduce**
Steps to reproduce the behavior:
Install react-native-ble-manager, run `pod install`, deploy app to device or simulator.

**Expected behavior**

**Smartphone (please complete the following information):**
 - Device: iPhone 16 Pro (Simulator)
 - OS: iOS 18
 - react-native-ble-manager version: v12.1.4
 - react-native version: 0.78.1

**Additional context**
@callstackincubator\polygen is also installed. However, **no crash occurred** before installing BleManager.

lucaswitch, 2025-03-21:
@thompsonnoah i dont know if this help you with the issue but i would add .start method outside react component lifecycle and would use only one instance of blemanager.

thompsonnoah, 2025-03-22:
Hi @lucaswitch,

Thank you for the tips! A couple of comments:

1. I noticed in the example app that the `start` method is in a `useEffect` call, so I mimicked that as well in my own app. Is this not the recommended approach?
2. I only use the `BleManager` once in the app, in the start method, which is imported like this: `import BleManager from 'react-native-ble-manager';`

If you could provide me with any additional guidance, that would be really appreciated 😃

lucaswitch, 2025-03-24:
> Hi [@lucaswitch](https://github.com/lucaswitch),
>
> Thank you for the tips! A couple of comments:
>
> 1. I noticed in the example app that the `start` method is in a `useEffect` call, so I mimicked that [...]

What i would do:
## Double check if new arch) are enabled
Probably that is not the issue as i can see on your provided image that the turbo modules call are being executed.

## Ensure that ".start" is only called when the app starts(maybe)
It might not be the cause of the issue, but it's worth checking. In our example, we use useEffect(), which works for demonstration purposes. However, managing it outside the component lifecycle—such as using a context provider, calling the "start method" there, and accessing it with useContext(Provider)—can be easier.

Alternatively, we can keep the BleManager instance inside useEffect() and ensure the "start method" runs only once by storing the state in useRef. However, this isn’t always reliable since the component could unmount and remount, creating a new instance and resetting the ref.

Personally, I find it much easier to handle it the way I suggested. Also, React's strict mode could be a factor, as it forces components to render at least twice in development mode.

Example:
[image]

If this continue happening i recommend you paste the full stacktrace so we can try to reproduce it somehow on the current workstation that we have.

Best regards

thompsonnoah, 2025-03-24:
Hi @lucaswitch,

Thank you so much for the assistance! I tried doing as you suggested, by creating a `Context` like this:

```
import {createContext} from 'react';
import BleManager from 'react-native-ble-manager';

BleManager.start({showAlert: false}).then(() => console.log('BleManager init'));

const BleContext = createContext(BleManager);

export default BleContext;
```

Then I wrap the `App` in the `Provider`:

```
import {AppRegistry} from 'react-native';
import App from './App';
import {name as appName} from './app.json';
import BleContext from './src/BleContext';

AppRegistry.registerComponent(appName, () => <BleContext.Provider>
  <App/>
</BleContext.Provider>);

```

Unfortunately, there is still a crash. Please let me know if you see any issues with my implementation. The stacktrace is listed here:

```
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x48caf180c)
    frame #0: 0x000000048caf180c
    frame #1: 0x0000000194f6f208 libswiftCore.dylib`(anonymous namespace)::DecodedMetadataBuilder::createBoundGenericType(swift::TargetContextDescriptor<swift::InProcess> const*, __swift::__runtime::llvm::ArrayRef<swift::MetadataOrPack>, swift::MetadataOrPack) const + 452
    frame #2: 0x0000000194f6a420 libswiftCore.dylib`swift::Demangle::__runtime::TypeDecoder<(anonymous namespace)::DecodedMetadataBuilder>::decodeMangledType(swift::Demangle::__runtime::Node*, unsigned int, bool) + 2104
    frame #3: 0x0000000194f6c3ec libswiftCore.dylib`swift::Demangle::__runtime::TypeDecoder<(anonymous namespace)::DecodedMetadataBuilder>::decodeMangledType(swift::Demangle::__runtime::Node*, unsigned int, bool) + 10244
    frame #4: 0x0000000194f676a4 libswiftCore.dylib`swift_getTypeByMangledNodeImpl(swift::MetadataRequest, swift::Demangle::__runtime::Demangler&, swift::Demangle::__runtime::Node*, void const* const*, std::__1::function<void const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 588
    frame #5: 0x0000000194f673e0 libswiftCore.dylib`swift_getTypeByMangledNode + 552
    frame #6: 0x0000000194f67be8 libswiftCore.dylib`swift_getTypeByMangledNameImpl(swift::MetadataRequest, __swift::__runtime::llvm::StringRef, void const* const*, std::__1::function<void const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 1008
    frame #7: 0x0000000194f62b64 libswiftCore.dylib`swift_getTypeByMangledName + 552
    frame #8: 0x0000000194f62e6c libswiftCore.dylib`swift_getTypeByMangledNameInContextImpl(char const*, unsigned long, swift::TargetContextDescriptor<swift::InProcess> const*, void const* const*) + 140
    frame #9: 0x000000038ca732d4 WasmTest.debug.dylib`__swift_instantiateConcreteTypeFromMangledName at <compiler-generated>:0
  * frame #10: 0x000000038caf6130 WasmTest.debug.dylib`SwiftBleManager.init(bleManager=0x0000600000c6d170) at SwiftBleManager.swift:37:23
    frame #11: 0x000000038caf6904 WasmTest.debug.dylib`@objc SwiftBleManager.init(bleManager:) at <compiler-generated>:0
    frame #12: 0x000000038cae760c WasmTest.debug.dylib`-[BleManager init](self=0x0000600000c6d170, _cmd="init") at BleManager.mm:27:25
    frame #13: 0x000000038cfd8ed4 WasmTest.debug.dylib`RCTAppSetupDefaultModuleFromClass(moduleClass=BleManager, dependencyProvider=0x0000600000c68720) at RCTAppSetupUtils.mm:112:10
    frame #14: 0x000000038cfdf568 WasmTest.debug.dylib`-[RCTReactNativeFactory getModuleInstanceFromClass:](self=0x0000600001770700, _cmd="getModuleInstanceFromClass:", moduleClass=BleManager) at RCTReactNativeFactory.mm:156:10
    frame #15: 0x000000038d1cd3fc WasmTest.debug.dylib`-[RCTInstance getModuleInstanceFromClass:](self=0x00006000037f5140, _cmd="getModuleInstanceFromClass:", moduleClass=BleManager) at RCTInstance.mm:193:31
    frame #16: 0x000000038cfaa4f4 WasmTest.debug.dylib`-[RCTTurboModuleManager _getModuleInstanceFromClass:](self=0x000000010c2ca730, _cmd="_getModuleInstanceFromClass:", moduleClass=BleManager) at RCTTurboModuleManager.mm:841:35
    frame #17: 0x000000038cfa8f78 WasmTest.debug.dylib`-[RCTTurboModuleManager _createAndSetUpObjCModule:moduleName:moduleId:](self=0x000000010c2ca730, _cmd="_createAndSetUpObjCModule:moduleName:moduleId:", moduleClass=BleManager, moduleName="BleManager", moduleId=89) at RCTTurboModuleManager.mm:639:12
    frame #18: 0x000000038cfa8d98 WasmTest.debug.dylib`__71-[RCTTurboModuleManager _provideObjCModule:moduleHolder:shouldPerfLog:]_block_invoke(.block_descriptor=0x0000600002142e40) at RCTTurboModuleManager.mm:572:18
    frame #19: 0x000000038cc194f4 WasmTest.debug.dylib`__RCTUnsafeExecuteOnMainQueueSync_block_invoke(.block_descriptor=0x000000016f81ff40) at RCTUtils.m:295:7
    frame #20: 0x0000000100a2a5d0 libdispatch.dylib`_dispatch_client_callout + 16
    frame #21: 0x0000000100a3b738 libdispatch.dylib`_dispatch_async_and_wait_invoke + 192
    frame #22: 0x0000000100a2a5d0 libdispatch.dylib`_dispatch_client_callout + 16
    frame #23: 0x0000000100a3a41c libdispatch.dylib`_dispatch_main_queue_drain + 1228
    frame #24: 0x0000000100a39f40 libdispatch.dylib`_dispatch_main_queue_callback_4CF + 40
    frame #25: 0x000000018041cdac CoreFoundation`__CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
    frame #26: 0x0000000180417308 CoreFoundation`__CFRunLoopRun + 1944
[... cut ...]
    frame #31: 0x0000000184f0a334 UIKitCore`UIKit.UIApplicationMain(Swift.Int32, Swift.Optional<Swift.UnsafeMutablePointer<Swift.UnsafeMutablePointer<Swift.Int8>>>, Swift.Optional<Swift.String>, Swift.Optional<Swift.String>) -> Swift.Int32 + 100
    frame #32: 0x000000038ca73c38 WasmTest.debug.dylib`static UIApplicationDelegate.main() at <compiler-generated>:0
    frame #33: 0x000000038ca73bb0 WasmTest.debug.dylib`static AppDelegate.$main() at <compiler-generated>:0
    frame #34: 0x000000038ca73cb4 WasmTest.debug.dylib`main at AppDelegate.swift:7:7
    frame #35: 0x0000000100c69410 dyld_sim`start_sim + 20
    frame #36: 0x0000000100b46274 dyld`start + 2840
```

thompsonnoah, 2025-03-25:
Hi again @lucaswitch,

I decided to give a try at solving the issue myself, and made some changes locally to see if I could get something working. I changed this line:

https://github.com/innoveit/react-native-ble-manager/blob/5d325727dee780ca08a09c7f87c4c9b74bd7c843/ios/SwiftBleManager.swift#L37

To this:

```
peripherals = Dictionary<String, Peripheral>()
```

And to my *complete and utter suprise*, no more crash! 🎉

I'm going to keep this issue open until I've tested the library to make sure everything still works. I am by no means an expert Swift developer, so I'd really like to hear your thoughts on this fix and if I should open up a PR.

thompsonnoah, 2025-03-25:
Hi again @lucaswitch,

Unfortunately, this fix introduces a lot more issues than it solves. It crashes again when doing a scan and exhibits odd behavior (more EXC_BAD_ACCESS crashes). If you have any updates for me on a more permanent fix, I would really appreciate it!

Thank you!

Noah

thompsonnoah, 2025-03-26:
Hi @lucaswitch,

Sure! I'll try to reproduce it in the sample app.

thompsonnoah, 2025-05-14:
Hi @lucaswitch,

I think this issue isn't a problem anymore. It might've been a conflict with another library, or something of that sort. Closing this for now, will re-open if it happens again.

Thanks,

Noah

closed 2025-05-14 by thompsonnoah