innoveit/react-native-ble-manager #1350: where is the event manager doc for version 11.5.7?

diffent, 2025-03-30:
in your version 12 release notes it says this:

"Event handling has changed, check the [documentation](https://innoveit.github.io/react-native-ble-manager/events/) to find the differences."

we are stuck at ble manager 11.5.7 for now due to using a slightly older react-native version, can you tell us where to find the event handler documentation for this older version?

thanks!

lucaswitch, 2025-03-31:
You can use the previous implementation which is bleManager.addEventListener('onStopScan') for example.

diffent, 2025-03-31:
hello thanks,

if anyone else is looking for this, we found in the old version that some of the events need a string 'BleManager' prefix like this:

 BleManager.addListener('BleManagerDidUpdateValueForCharacteristic', (event) => {
                        console.log("event", event)
})

the error messages clued us in

just getting started on this module, so don't know if this is universal for all events

CptFabulouso, 2025-05-07:
It's unfortunate that docs for v11 are not accessible anymore. Anyway these are the supported events as defined in the ios v11 project:

"BleManagerDidUpdateValueForCharacteristic", 
"BleManagerStopScan", 
"BleManagerDiscoverPeripheral", 
"BleManagerConnectPeripheral", 
"BleManagerDisconnectPeripheral", 
"BleManagerDidUpdateState", 
"BleManagerCentralManagerWillRestoreState", 
"BleManagerDidUpdateNotificationStateFor"

diffent, 2025-05-07:
Thank you.  Very helpful.