innoveit/react-native-ble-manager #1359: Why can't I scan iBeacon on Android, but I can on iOS?

yiluzhang, 2025-04-26:
Device info, from nRF connect App:

[image]

"react-native-ble-manager": "11.5.7"

```tsx
import React, { useEffect, useState } from 'react';
import { Button, PermissionsAndroid, Platform, StyleSheet, Text, TextInput, View } from 'react-native';
import BleManager, { BleScanMode, Peripheral } from 'react-native-ble-manager';
import { FlashList } from '@shopify/flash-list';
import { Buffer } from 'buffer';

const BluetoothScanScreen: React.FC = () => {
  const [scanning, setScanning] = useState(false);
  const [scanResults, setScanResults] = useState<Peripheral[]>([]);
  const [searchText, setSearchText] = useState('');
  const [excludeNameText, setExcludeNameText] = useState('');
  const [filteredResults, setFilteredResults] = useState<Peripheral[]>([]);

  useEffect(() => {
    BleManager.start({ showAlert: false });

    const onPeripheralDiscovered = (peripheral: Peripheral) => {
      setScanResults(prevResults => {
        const exists = prevResults.some(device => device.id === peripheral.id);

        if (!exists) {
          return [...prevResults, peripheral];
        }

        return prevResults;
      });
    };

    const listener = BleManager.addListener('BleManagerDiscoverPeripheral', onPeripheralDiscovered);

    return () => {
      listener.remove();
    };
  }, []);

  useEffect(() => {
    let filtered = scanResults.filter(device => {
      const name = (device.name || '').toLocaleLowerCase();
      const localName = (device.advertising.localName || '').toLocaleLowerCase();

      const includeCondition =
        !searchText || name.includes(searchText.toLowerCase()) || localName.includes(searchText.toLowerCase());

      const excludeKeywords = excludeNameText
        .toLowerCase()
        .split(' ')
        .map(k => k.trim())
        .filter(k => k !== '');

      const excludeCondition =
        excludeKeywords.length === 0 || !excludeKeywords.some(k => name.includes(k) || localName.includes(k));

      return includeCondition && excludeCondition;
    });

    filtered = filtered.sort((a, b) => b.rssi - a.rssi);
    setFilteredResults(filtered);
  }, [scanResults, searchText, excludeNameText]);

  const requestBluetoothPermissions = async (): Promise<boolean> => {
    if (Platform.OS === 'android') {
      const granted = await PermissionsAndroid.requestMultiple([
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
        PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADVERTISE,
      ]);

      return (
        granted['android.permission.ACCESS_FINE_LOCATION'] === PermissionsAndroid.RESULTS.GRANTED &&
        granted['android.permission.BLUETOOTH_SCAN'] === PermissionsAndroid.RESULTS.GRANTED &&
        granted['android.permission.BLUETOOTH_CONNECT'] === PermissionsAndroid.RESULTS.GRANTED
      );
    }

    return true;
  };

  const startScan = async () => {
    const hasPermissions = await requestBluetoothPermissions();

    if (hasPermissions) {
      setScanning(true);
      setScanResults([]);
      BleManager.scan([], 15, true, { scanMode: BleScanMode.LowLatency });
    }
  };

  const stopScan = () => {
    if (scanning) {
      BleManager.stopScan().then(() => {
        setScanning(false);
      });
    }
  };

  const renderField = (label: string, value: any) => {
    if (value === null || value === undefined || value === '') {
      return null;
    }

    let displayValue = value;

    if (typeof value === 'object' && value.CDVType === 'ArrayBuffer' && value.bytes) {
      displayValue = Buffer.from(value.bytes).toString('hex').toUpperCase();
    } else if (typeof value === 'object') {
      displayValue = JSON.stringify(value);
    }

    return (
      <Text style={styles.fieldText}>
        {label}: {displayValue}
      </Text>
    );
  };

  const renderItem = ({ item }: { item: Peripheral }) => {
    return (
      <View style={styles.listItem}>
        {renderField('ID', item.id)}
        {renderField('Name', item.name || item.advertising.localName)}
        {renderField('RSSI', item.rssi)}
        {renderField('UUIDs', item.advertising.serviceUUIDs)}
        {renderField('Raw Data', item.advertising.rawData)}
        {renderField('Manufacturer Data', item.advertising.manufacturerRawData)}
        {renderField('Service Data Raw', item.advertising.serviceData)}
        {Object.keys(item.advertising.serviceData || {}).map(key =>
          renderField(`${key}`, item.advertising.serviceData?.[key]),
        )}
      </View>
    );
  };

  return (
    <View style={styles.container}>
      <TextInput style={styles.searchInput} placeholder="name" value={searchText} onChangeText={setSearchText} />
      <TextInput
        style={[styles.searchInput]}
        placeholder="exclude name"
        value={excludeNameText}
        onChangeText={setExcludeNameText}
      />
      <View style={styles.buttonContainer}>
        <Button title={scanning ? 'Start scan' : 'Stop scan'} onPress={scanning ? stopScan : startScan} />
      </View>
      <FlashList
        data={filteredResults}
        renderItem={renderItem}
        keyExtractor={item => item.id}
        estimatedItemSize={100}
      />
    </View>
  );
};

const styles = StyleSheet.create({
  container: {
    flex: 1,
    padding: 20,
  },
  searchInput: {
    height: 40,
    borderColor: 'gray',
    borderWidth: 1,
    marginBottom: 10,
    paddingHorizontal: 10,
    borderRadius: 8,
  },
  buttonContainer: {
    marginBottom: 10,
  },
  listItem: {
    padding: 12,
    marginBottom: 12,
    backgroundColor: '#f9f9f9',
    borderRadius: 8,
    elevation: 2,
    minHeight: 60,
  },
  fieldText: {
    marginBottom: 2,
    fontSize: 14,
    color: '#333',
  },
});

export default BluetoothScanScreen;
```

```xml
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
 <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
```

iOS scan result
[image]

Android scan result does not include this!

blitzsports, 2025-05-04:
My suggestion would be to try and roll back to 11.5.5 and see if that solves your problem. I had a similar problem that I've nailed down to occurring in 11.5.6 and beyond.

burak-sungur-it, 2025-08-03:
You need to add this to your Android manifest:
<uses-permission android:name=”android.permission.BLUETOOTH_SCAN” android:usesPermissionFlags=”neverForLocation”/>