innoveit/react-native-ble-manager #1361: [iOS] Display alert for every `start()`

CptFabulouso, 2025-05-07:
## What I want

I want to be able to use the `start({ showAlert: true, queueIdentifierKey: 'key', restoreIdentifierKey: 'key' } )` to always prompt the user to enable bluetooth if it is off. 
In the app I check for the BT state, if it is off, I show button to enable it.

## Problem

The prompt only displays once, since there is [this code](https://github.com/innoveit/react-native-ble-manager/blob/master/ios/SwiftBleManager.swift#L208) which prevents the reinitialization of CBCentralManager and therefore re-displaying of the alert.

## Question

What do we lose if the `CBCentralManager` would be always initialized?  Would that make the whole CBCentralManagerOptionRestoreIdentifierKey obsolete? I quite don't fully understand the whole purpose of that flag. Right now it's either omit the `restoreIdentifierKey` or only display prompt once. Can we have both?
I don't want to display some kind of modal instructing user to go to settings and enable BE.

marcosinigaglia, 2025-05-12:
It is an iOS instruction to initialize only one `CBCentralManager` per app. Initializing multiple times can lead to unexpected behavior. Also from a usability point of view I suspect that showing the popup multiple times would be against Apple's guidelines and you would risk the app being rejected.

CptFabulouso, 2025-05-12:
@marcosinigaglia Code-wise (on native) if you don't provide `restoreIdentifierKey` and call start multiple times, then `CBCentralManager` is initialized multiple times. So if the native code is supposed to prevent that, then it doesn't in that case. I've looked at some iOS docs [here](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/PerformingCommonCentralRoleTasks/PerformingCommonCentralRoleTasks.html#//apple_ref/doc/uid/TP40013257-CH3-SW2) and [here](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/init()) and I didn't find any instruction or recommendation saying to initialize it only once. Any chance you could link me where it's stated?

If the alert would popup after pressing a button with label "Enable bluetooth", then I think it should be fine. I mean the context is user trying to connect to BT device and we want to give him a handy way to enable it, we're not popping it on him out of nowhere.

marcosinigaglia, 2025-05-14:
There is no specific warning because you can initialize `CBCentralManager` multiple times within the same app if you have for example external libraries or other special cases. In this case within the library I don't really see the need to allow it to be initialized multiple times, especially knowing how Apple thinks in terms of resources and user interaction.
You can ask in the Apple forum if this procedure can be considered valid or provide some other example/article where they suggest this mechanism.