innoveit/react-native-ble-manager #1372: onDiscoverPeripheral not working on iOS

sohaibbinmohsin, 2025-07-01:
I am not getting any peripherals when testing my app in iOS. All permissions are given but still after 20 seconds, i get the error saying no devices found. What am i doing wrong here?

Bluetooth permission status: granted
Location permission status: granted

```
  const startRealScan = () => {
      let onDiscoverListener: any;
      addDebugLog('Starting real Bluetooth scan...');

      BleManager.start({showAlert: true}).then(() => {
        addDebugLog('BLE Manager started successfully');
      });

      BleManager.checkState().then(state =>
        console.log(`current BLE state = '${state}'.`),
      );
      // Register the callback for peripheral discovery
      onDiscoverListener = BleManager.onDiscoverPeripheral(
        (peripheral: any) => {
          addDebugLog(
            `Found: ${peripheral.name || 'Unknown'} (${peripheral.rssi} dBm)`,
          );

          // Determine device type based on advertising data
          const getDeviceType = (peripheral: any) => {
            const {advertising} = peripheral;

            // Check for specific service UUIDs that indicate device type
            if (advertising?.serviceUUIDs) {
              const services = advertising.serviceUUIDs.map((uuid: string) =>
                uuid.toLowerCase(),
              );

              // Common service UUIDs for different device types
              if (services.some((uuid: string) => uuid.includes('180d')))
                return 'Fitness Tracker';
              if (services.some((uuid: string) => uuid.includes('180f')))
                return 'Battery Service';
              if (services.some((uuid: string) => uuid.includes('1812')))
                return 'HID Device';
              if (services.some((uuid: string) => uuid.includes('110a')))
                return 'Audio Device';
              if (services.some((uuid: string) => uuid.includes('110e')))
                return 'Phone';
              if (services.some((uuid: string) => uuid.includes('110d')))
                return 'Computer';
            }

            // Check manufacturer data for device type hints
            if (advertising?.manufacturerData) {
              const manufacturerId = Object.keys(
                advertising.manufacturerData,
              )[0];
              // Apple devices
              if (manufacturerId === '76') return 'Apple Device';
              // Samsung devices
              if (manufacturerId === '15') return 'Samsung Device';
              // Google devices
              if (manufacturerId === '224') return 'Google Device';
            }

            // Check device name for hints
            const name = peripheral.name?.toLowerCase() || '';
            if (name.includes('iphone') || name.includes('ipad'))
              return 'Apple Device';
            if (name.includes('samsung')) return 'Samsung Device';
            if (name.includes('jsn') || name.includes('huawei'))
              return 'Huawei Device';
            if (name.includes('macbook') || name.includes('mac'))
              return 'Computer';
            if (name.includes('laptop')) return 'Computer';
            if (name.includes('watch')) return 'Smartwatch';
            if (name.includes('headphone') || name.includes('buds'))
              return 'Audio Device';
            if (name.includes('speaker')) return 'Speaker';
            if (name.includes('fitbit') || name.includes('garmin'))
              return 'Fitness Tracker';

            return 'Unknown Device';
          };

          // Generate a friendly name
          const getDeviceName = (peripheral: any) => {
            if (peripheral.name && peripheral.name.trim()) {
              return peripheral.name;
            }

            if (peripheral.advertising?.localName) {
              return peripheral.advertising.localName;
            }

            const deviceType = getDeviceType(peripheral);
            const shortId = peripheral.id.slice(-4);
            return `${deviceType} (${shortId})`;
          };

          const deviceName = getDeviceName(peripheral);
          const deviceType = getDeviceType(peripheral);
          const rssi = peripheral.rssi || -100;

          // Update discovered devices list
          setDiscoveredDevices(prev => {
            const existingIndex = prev.findIndex(d => d.id === peripheral.id);
            if (existingIndex >= 0) {
              // Update existing device with new RSSI
              const updated = [...prev];
              updated[existingIndex] = {
                ...updated[existingIndex],
                rssi: rssi,
                lastSeen: new Date(),
                name: deviceName,
                deviceType: deviceType,
              };
              addDebugLog(`Updated: ${deviceName} (${rssi} dBm)`);
              return updated;
            } else {
              // Add new device
              const newDevices = [
                ...prev,
                {
                  id: peripheral.id,
                  name: deviceName,
                  rssi: rssi,
                  lastSeen: new Date(),
                  deviceType: deviceType,
                },
              ];
              // Update the ref with current count
              discoveredDevicesCountRef.current = newDevices.length;
              addDebugLog(
                `Added: ${deviceName} (${rssi} dBm) - Total: ${newDevices.length}`,
              );
              return newDevices;
            }
          });

          // Update highest RSSI for the meter
          if (rssi > highestRssi && rssi > -82) {
            highestRssi = rssi;
            setFinalRssi(highestRssi);
            addDebugLog(`New highest RSSI: ${highestRssi}`);
          }

          // Update device lines for chart (only if RSSI is reasonable)
          if (rssi > -80) {
            setDeviceLines(prev => {
              let found = false;
              const updated = prev.map(line => {
                if (line.id === peripheral.id) {
                  found = true;
                  return {
                    ...line,
                    data: [
                      ...line.data,
                      {
                        value: rssi,
                        dataPointText: String(rssi),
                      },
                    ].slice(-30), // keep last 30 points
                  };
                }
                return line;
              });
              if (!found) {
                // Assign a color based on RSSI value
                const color =
                  rssi >= -56 ? 'red' : rssi >= -71 ? 'orange' : 'green';
                addDebugLog(`Added to chart: ${deviceName} (${rssi} dBm)`);
                return [
                  ...updated,
                  {
                    id: peripheral.id,
                    name: deviceName,
                    color,
                    data: [
                      {
                        value: rssi,
                        dataPointText: String(rssi),
                      },
                    ],
                  },
                ];
              }
              return updated;
            });
          }
        },
      );

      BleManager.scan([], 0, true);
      isScanningRef.current = true;

      // Stop scan after 20 seconds
      const timeout = setTimeout(() => {
        addDebugLog('Scan timeout reached. Stopping scan...');
        if (isScanningRef.current) {
          BleManager.stopScan();
        }
        isScanningRef.current = false;
        setScanState('done');

        if (onDiscoverListener) {
          onDiscoverListener.remove();
        }

        // Check if no devices were found using the ref
        const deviceCount = discoveredDevicesCountRef.current;
        addDebugLog(`Scan finished. Total devices found: ${deviceCount}`);

        if (deviceCount === 0) {
          addDebugLog('No devices found, showing alert');
          Alert.alert(
            'No Devices Found',
            'No Bluetooth devices were discovered during the scan. This could be because:\n\n• No devices are nearby\n• Bluetooth is disabled on nearby devices\n• Devices are not advertising\n• Location permission is required for Bluetooth scanning\n\nTry moving to a different location or ensure Bluetooth is enabled on nearby devices.',
            [
              {
                text: 'OK',
                onPress: () => navigation.navigate('MainTabs'),
              },
            ],
          );
        } else {
          addDebugLog(`✅ ${deviceCount} devices found successfully`);
          // Show success alert
          Alert.alert(
            'Scan Complete',
            `Successfully discovered ${deviceCount} Bluetooth device(s)!`,
            [{text: 'OK'}],
          );
        }
      }, 20000);

      setScanTimeout(timeout);
    };
```

roniemartinez, 2025-07-10:
We are getting the same issue here on iOS (intermittently, probably works 1x in 20 tries or even less). Our code works very good on Android! (We are scanning a BLE thermal printer). 

`checkState()` returns `on` and there's probably less "configuration" for iOS than Android (close to 1 or 2) when using `start()` and `scan()` and we tried them all so not sure about the configuration problem.
Thanks to anyone who can figure this out.

Daniel-Reiling, 2025-07-24:
We are facing the same issue while working to migrate our older react native 0.72 app to a new Expo 53 + RN 0.79 foundation. During that process I have tried react-native-ble-manager 11.0.0 and 12.0.0 and both result in no devices detected when scanning (with the same code that works perfectly on RN0.72 and ble-manager 10.1.3). 

As imagined this is a huge roadblock for updating our app. If there is any additional help we can provide to help resolve the issue we will to avoid changing our underlying BLE implementation.

marcosinigaglia, 2025-07-25:
> > Are you using the new architecture?
> 
> Yes, we are. Here is some addition information below. I am also working on creating a minimal example now.
> 
> Versions:
> 
> ```
> "expo": "~53.0.11",
> "react": "19.0.0",
> "react-native": "0.79.3",
> 
> DEV: 
> "@types/react": "~19.0.10",
> "@types/react-native": "^0.73.0",
> "typescript": "~5.8.3"
> ```
> 
> And app.config.ts has:
> 
> ```
> newArchEnabled: true,
> ```

Can you try the example project? I tested now and it's working.

Loovery, 2025-07-26:
Maybe this will help. The problem appeared after updating RN from version 0.76 (old arch) > 0.80 (new arch). And ble-manager update from version 11.X > 12.1.5

Loovery, 2025-07-26:
Works for me. Documentation has changed for events (Used the old way via addListener, instead of BleManager.on...), noticed when I started prototyping.

mathanraj-oben, 2025-08-19:
any solution? i'm facing same issue

react-native: 0.76.7
iOS: 18.5

marcosinigaglia, 2025-08-19:
> any solution? i'm facing same issue
> 
> react-native: 0.76.7 iOS: 18.5

There are no known issues. Have you tried the sample app?

closed 2025-11-08 by marcosinigaglia